vlachoudis/bCNC #668: laser etching

dodolbysuround, 2017-10-07:
Hi every one,
I'm wondering which would be the straighter way for marking a part with letters composed of several little dots rather than 1 continuous trace.
Is there any existing plug or do I need to draw my dot composed letters "manualy" in DXF?
Thanks for your answers.

vlachoudis, 2017-10-07:
Maybe you can use the driller plugin of @Effer

Effer, 2017-10-07:
What about a Dot Font with the Text Plugin?
[image]

Tested with one this : https://www.1001freefonts.com/dotted-fonts.php

dodolbysuround, 2017-10-07:
Thank you guys for your help. 
I'm trying Vlachoudis method (ie Effer's Driller plugin but althoug I checked the élaser cuter" option in "configuration" tab,  I get Z0, Z5, Z0, Z5... instead of M3, M5, M3, M5, whe I execute the driller prugguin.
Any idea?

Effer, 2017-10-08:
I think I can try to patch driller to generate M3/M5 commands when in lsaer mode. It's a small change.

vlachoudis, 2017-10-08:
@effer it is better to use the zentry/zexit from CNC.py

Effer, 2017-10-08:
@vlachoudis , just check, it already use it... so probably is only a matter of restart bCNC to allow settings take effect,

Effer, 2017-10-08:
This should work: https://github.com/vlachoudis/bCNC/pull/672
Please report @dodolbysuround

dodolbysuround, 2017-10-08:
thank you sooo much! This works theoretically. I'll try it in a practical way on a real piece of material tomorrow

dodolbysuround, 2017-10-09:
thank you Filoppo, This is perfect.
Just One thing: there is a problem when I try to re locate the "driller" related g-code with the pre defined "move" fonctions (top right, center, top left, ...) While I try do do so, G-code is moved away faaaaaaaar away from the original place.
any explanation for that?

dodolbysuround, 2017-10-09:
another question:
is there any way on bCNC to sweep at constant speed the overall area and throw one laser shoot when flying over parts of the pictures that need to be engraved?
pyrograph plugin works almost like this but instead of variing feed rate, I'm thinking about turning on and off the laser.

lalo-uy, 2017-10-09:
>
> While I try do do so, G-code is moved away faaaaaaaar away from the
> original place.
> any explanation for that?

Check it its not moved in inches insted of mm

dodolbysuround, 2017-10-11:
no, unit are set to mm