vlachoudis/bCNC #670: Problems with python-opencv 2.x

chaotix-, 2017-10-07:
A recent change in Camera.py gets some CV properties, e.g. cv.CAP_PROP_FRAME_HEIGHT

This property is not set in CV 2.x. There the correct name is:
cv.cv.CV_CAP_PROP_FRAME_HEIGHT

This throws an exception and stops bCNC from starting.

vlachoudis, 2017-10-08:
I've put an exception to handle both formats. Could you try it now?

chaotix-, 2017-10-08:
Probably works fine. "Probably" because I don't know what these properties are supposed to do. I did not find any reference to 'contrast', 'exposure', etc. in the GUI or the rest of the code. But bCNC starts up fine.

It's probably nit-picking, but I would change 'except:' to 'except AttributeError:'. It work's either way, of course.

vlachoudis, 2017-10-20:
ok done

closed 2017-10-20 by vlachoudis