vlachoudis/bCNC #673: Noob Question

MikeMTM, 2017-10-08:
Hi bCNC Community,
I recently finished building my first GRBL based cnc machine. I've spent so much time on the build and I'm finally to the point where I can run some gcode. I designed a very basic profile cut in Fusion 360 as my first job, sort of like a way to tune the machine. So I move the spindle to the piece of stock in the middle of my table and zero the Wpos. I then load the gcode and press run / start. My Z axis moves up to its "0" position and trips the limit switch. That's as far as I get. I can see that the movement slows down as it's approaching the switch so it's not trying to go past it. I've literally been trying to figure this out all day. I wasn't going to ask but I'm out of ideas. The error I get is: ALARM 1; Last Error: S800M3. Thanks!

chaotix-, 2017-10-09:
I don't understand this. You set Z0 as the top of your piece of stock? Then "moves up to its "0" position" does not make any sense.

Did you press "Z = 0" when the end mill was on top of your stock?

I have never used Fusion 360, so I don't know how it handles the coordinates. But there is no implicit need that the top of your workpiece is Z0. It could be anything. What is the GCode that was executed when it hit your hard limit? (You can see the past gcodes (and the queued ones) in the terminal tab.)

If you switch the view to XZ or YZ you can see the Z range of your file in the view and you also see the current position of your end mill. Does this look alright? When you place the end mill on top of the workpiece does this fit with the first depth the file tries to cut?

If you have a very limited Z range and your stock piece is high, sometimes the "safe height" (or whatever it is called in Fusion 360) might be outside your Z working area. Then you need to reduce this height in Fusion 360.

vlachoudis, 2017-10-09:
Be carefull, Fusion 360 adds a G28 (return to ref. position) at the begging of the code.
If you haven't set the G28 on your machine it will go to the 0 (machine position) and can potentially
trip your limit switch.
Either you remove the G28 or you properly set the G28 ref.

MikeMTM, 2017-10-09:
YES! That was the problem. I finally found some info regarding the G28 reference point last night. You can either move the spindle to a chosen spot and issue G28.1 to set it or you can completely disable it in the post processor. I ended up disabling it and everything works now. Thanks for your help. At the time I wasn't sure if it was some GRBL specific setting I missed.

closed 2017-10-09 by MikeMTM