vlachoudis/bCNC #674: Set zero function doesnt work

Nikoju, 2017-10-10:
Hi, I've got the following issue.
when I want to set, for example, the z-axis to zero or a defined offset every single axis gets set to zero.
Does anyone know what I'doing wrong?

vlachoudis, 2017-10-20:
Its hard to know what it is going on. Can you try typing 0 in one of the axis and pressing enter and check?

Nikoju, 2017-10-21:
Hi thanks for your response, When I set it to zero and press enter it also set every single axis to zero. 
But It turned out that I had a bad version of grbl installed to my Arduino. I reinstalled it via the Arduino IDE and now it works fine but still, there is one problem. When I press the pause button the CNC stops like it's supposed to be but I cant restart it. It would be great if you could help me out with this problem.
thanks!

vlachoudis, 2017-10-30:
You should press Pause again to continue

closed 2018-10-29 by Harvie