vlachoudis/bCNC #675: 3D scanner

Nossolov, 2017-10-11:
Hi every one.
Just an idea.
This week, i try autoleveling et it works pretty well.
I am not good enough to do this, but i'm wondering if it's very hard to adapt autoleveling process in 3D scanner. A Z probe, an area, a X,Y grid : exactly what is needed. A file at the end with Z measured for each (x,y) point, and you have a 3D scanner.

dodolbysuround, 2017-10-11:
I did this few years ago but not with bCNC. mp me if you want to discuss this.

Effer, 2017-10-11:
@Nossolov if you run an autolevel task, you can later save probe data in STL format.
see here: https://github.com/vlachoudis/bCNC/pull/142

Nossolov, 2017-10-11:
Thank's Effer. I have not see this function yet, but there are so many things to discover in bCNC...
I try this option asap. STL is fine.
Great : i have now a 3D scanner !

closed 2017-10-11 by Nossolov