vlachoudis/bCNC #676: another laser feature

dodolbysuround, 2017-10-11:
is there any way on bCNC to sweep at constant speed the overall area and
 throw one laser shoot when flying over parts of the pictures that need to
 be engraved?
 pyrograph plugin works almost like this but instead of variing feed rate,
 I'm thinking about turning on and off the laser.

Effer, 2017-10-11:
It could be add as option to pyrograph plugin. Sadly I can't test results, so not sure if I'll work on this.

dodolbysuround, 2017-10-11:
I can be your beta tester (-;

Effer, 2017-10-11:
Can you describe your setup? Laser model/power work area size. Pictures appreciated :)

dodolbysuround, 2017-10-11:
I'm using a 12 watts chines laser (like this one but more powerfull)
[image]
on a small custom CNCed milling maching (sorry, I have no picture right kno but I'll take some toomorow. the max work area size is around 300mm (x) 110mm (y)
Here is a picture of what I already did with your updated driller plugin and my laser.
[image]
it measures around 10 x 10 mm.

dodolbysuround, 2017-10-12:
here is the picture of my machine.
[image]

indyucoder, 2017-10-12:
Nice, look like it is able to engrave on metal.

dodolbysuround, 2017-10-12:
yes, I can easily replace the laser by a spindle depending on what I need to do. every screws has been replaced by ball screws

Effer, 2017-10-12:
Wow, that's a big machine!
So you'll like to sweep forward and back and fire the laser when brightness is under a settable value?
Does the laser have to stop when reaching the point? What if there are more points in line to draw?

dodolbysuround, 2017-10-12:
arf... not that big actually (-;
I have been thinking quite a lot about all that.
There are two problem I can see with pyrographe plugin used with a laser:
- given the laser is always on, the clearest  area is burnt anyway (not strongly but still) 
- due to acceleration and deceleration, boundary lines are not very neat.
to get reed of these problems, I suggest 2 modifications:
- when traveling over clear areas, G0 should be  used and laser should be OFF (M5)
- when traveling over dark area, G1 should be used with a predefined feed rate that could be adjusted with a parameter in the setting list) and laser should be ON (M3)

In my best dreams, the whole sweeping process should be done at max speed (G0) but I'm afraid the laser wont be powerful enough to burn properly the areas it will have to burn.

I suggest we work only in black and write in a first time. so laser power (Sxxxx) can keep constant during the whole sweeping process. However, this would be interesting to get this value as a parameter is the setting list.

To answer your 2 questions: I don't realy know yet... The thing is, from what I observed, the laser delivers more power when fired by pulse than if it is used to draw a line. Furthermore, depending on the engraves material, it seems the darkness is more homogeneous with several little dots rather than one continuous line. Then I may prefer firing the laser by pulses but if it is quicker and/or easier, for the beginning,  it can be steady ON if there is a long dark area to engrave.

Not sure everything is clear...???

dodolbysuround, 2017-10-22:
hey @Effer 
I was wondering if you found some time and inspiration to modify the pyrographe plug in?

Effer, 2017-10-23:
Hi @dodolbysuround , inspiration? maybe. time? not at all. :smile:
Hope I can found some spare time next days. 
Probably I'll just add an option to skip cutting/laser when calculated feed is over a desired value.

Effer, 2017-11-17:
Sorry @dodolbysuround for late response. I should avoid promise something next time 😭 .
Anyway here a test: if the calculated feed (in relation with image brightness) is over "Skipping feed" the path is skipped or laser is turned off (in laser mode).

[image]

**Beware I didn't test the g-code on machine**
Unzip this file on plugins folder:  [pyrograph2.zip](https://github.com/vlachoudis/bCNC/files/1483702/pyrograph2.zip)

If you make some test let me know (I've buy a small laser but it will take some time to come from east)

dodolbysuround, 2017-11-17:
Greaaaaaaaaaaaaaat! I'll try to give it a try in the week end!
but please, help me understanding some point:
- There are 2 differents "laser modes": one in grbl and one in bCNC.  isn't it?
- How do I activate bCNC laser mode? ? I managed to do this but I don't know how, and I'm getting troubles trying to set it back in normal mode.
- I guess this is in tool/config tab but my "CNC button is always grey...
- May I have to restart bCNC for this setting to be taken under acount?

Effer, 2017-11-18:
* Yes, 2 laser mode. One in the grbl latest firmware that introduce some interesting features (like modulating laser power against the feed rate) and on in bCNC.
* to configure laser mode in bCNC is simple a matter to tick a checkbox in Tool->Config->Laser Cutter:
[image]
**Then remember to restart bCNC**

I'm going by memory so I could be wrong but when this option is enable most of the plugins avoid to move Z axis and instead turn laser On/Off with M3/M5