vlachoudis/bCNC #679: UGLY GUI UGLY ICONS

masterleo, 2017-10-13:
Hello I just write here because I think the icon on the GUI are SOOO ugly, it really annoys a lot of peoples please work on the graphical design and layout of this features rich program !!!!

masterleo, 2017-10-13:
This program will not be successful if the design stay so ugly.

closed 2017-10-13 by vlachoudis

atrueresistance, 2017-10-13:
It's open source, feel free to modify them and send a pull request.

DaveDavenport, 2017-10-13:
>  it really annoys a lot of peoples 

I would like to see some proof of this. So far N=1. You are the first I know that is annoyed by how the interface looks.

@vlachoudis I am very happy with how bCNC works and looks.  It is functional and usable. I love the amazing work you are doing.

masterleo, 2017-10-27:
I say this because I run the Fabrication laboratory in France where a lot of people use my CNC... I have at least 15 people (professionals, hobby) that think that this is so ugly. When I see how fast you are closing this thread in comparison of the critical situation that this represents I understand why Bcnc is so poorly developed in comparison to linuxCNC. 
@DaveDavenport Thanks for your reply, I do absolutely not speak of function here. Maybe it work for you but by me 60 people pay to use my machine and as I said the GUI is NOT successful.

DaveDavenport, 2017-10-27:
@masterleo If the looks of icons in a gui determines if it is successful or not for a user, the user is using it wrong.  A lot of professional software has a 'functional' gui that is not pretty, but usable. (mentor graphics, cadance, etc.).

That said, i agree a lot of people judge on initial looks, and therefor a nicer icon can help. Luckely this is opensource, feel absolutely free to provide a merge request with nice looking icons.

I don't think it is desirable to keep on opening new issues (#686), please keep the discussion in here.

vlachoudis, 2017-10-27:
@masterleo the thread is closed due to the language used in your post. Nobody claimed that the icons or the GUI is nice or not, and rest assured that constructive criticism is always taken into account. As correctly @DaveDavenport mentioned, the project is opensource. Please do the effort to design the new icons as you like and make a pull request, or do not use at all the software. Until then the topic will remain closed.