vlachoudis/bCNC #680: gcode wont fully load

joshcarz, 2017-10-16:
I generated some Gcode using fusion 360 for generic GRBL and the file won't fully load on bCNC. Does it work for anyone else? Here is a link to download my Gcode

https://www.dropbox.com/s/gjj7lblww7fm7ir/speedo%20bearing%20carrier%20v2.nc?dl=0

joshcarz, 2017-10-16:
So I just tried and it does cut everything out. It just doesnt load everything in the GUI and doesnt calculate anything like time remaining after what it can load. Maybe there is a limit to how many lines it can load?

vlachoudis, 2017-10-20:
This is a repeating question. We should make a FAQ and put it there as first. It is not a loading issue but rather a plotting. Python is slow for rendering and there is a timeout for the display on the top-right corner of the canvas. Increase that so it can show all your code.

closed 2017-10-20 by vlachoudis

lalo-uy, 2017-10-23:
May be a popup msg warning that timeout have been reached

meysamam, 2020-01-11:
Set Timeout to max 120

nuxeh, 2020-09-07:
This has been bugging me for a while, often I had to look for another gcode sender when my gcode wouldn't load. Thanks @meysamam

What is the timeout for..?

Harvie, 2020-09-07:
canvas is too slow to paint all g-code paths in long g-code files. there is timeout so it does not get stuck forever.
only problem is that g-code is not graphicaly drawn to display. otherwise the code is loaded and can be streamed to machine.
only way to solve this is to implement native code (eg. written in C/C++) g-code canvas and reuse it in bCNC rather than drawing g-code line by line directly from python, which is terribly slow language.

nuxeh, 2020-09-07:
Indeed, makes perfect sense... Or, even better, Rust :)