vlachoudis/bCNC #684: Integrate A Machine Timer [Enhancement]

atrueresistance, 2017-10-19:
@vlachoudis If I'm looking to integrate a machining timer where would I go about it?

The idea here is since bCNC runs all the code and it keeps track of execution time is there a easy way to have it sum all the time up and store it in like the bcnc.ini file?

It would be nice to view how many hours ran on a spindle to anticipate replacing router brushes or determine what hours of the day a particular machine is seeing the most use.

vlachoudis, 2017-10-20:
Do you want accurate timing that the spindle is on or some rough idea, e.g. total execution time.
You could potentially use the time of the progressbar of the status. I have to give a look on how to extract the information and save it.

CyberHugger, 2017-10-20:
Assuming we're not using spindle control I guess the only way you would know is based on the execution time for the tool paths. Not sure if you can pick that out of the gcode to decide what to measure or if we'll need to have a configuration item somewhere to tell you.  It gets even more complicated for those who may use the same machine for multiple uses like CNC and Laser.  How would you know? It might be easier to just add an AC hour meter to your spindle.

atrueresistance, 2017-10-21:
That is what I was thinking, based off the status. I don't think it needs to be super accurate. The multiple device question is something to consider. I'm one of those people using it for CNC and laser, but just looking for a general hour usage.

Harvie, 2018-10-29:
I think this should be implemented in GRBL or VFD as some people might use various g-code senders on that machine.

closed 2018-10-29 by Harvie