vlachoudis/bCNC #688: Feature Request: Adding BLOCKS between Object and Layers

kworsfold, 2017-10-29:
I would like to start by saying great job, I am slowing learning the full potential of this application.

There are however a couple of minor irritations I have found.
1) When creating an object the code is all lowercase and no preceding Zero, i.e. G01.  This just looks tidier when manually editing.
2) When generating the code it comes out as 2 blocks a header and 1 object. There is no Footer and just the single block, so when trying to add things like the TAB there is no way to easily select the bottom layer.

Could you look into tidying up the G-code a little in line with other applications. Also an option to add BLOCKS between Objects and Layers when generating the code.

One trick I use which would be nice is, in the Editor, clicking on the movements between objects and adding the code manually i.e (Block-name: layer 1). The only problem with this is I then have to save the File and reload.... which is frustrating. There could also be the option to refresh the editor. 
Something like this would also be very useful with a single click to complete the whole proceedure.

Keep up the good work.......

vlachoudis, 2017-10-30:
1. I am sorry, this was my personal taste :) I don't like capitals, I find small letters to be easier to distinguish from the numbers z12 or Z12. Maybe we can add an option on the formatting.
2. When I insert manually the code the header is not written. The header and footer should be generated when you import a dxf drawing or you automatically generate. Can you give me a step by step example to understand what you mean.

Also here can you give an example how and what BLOCKS to add between objects/layers?

Normally any change you would do in the editor it should automatically refresh the canvas. If not then there is some problem somewhere.

kworsfold, 2017-10-30:
It is a prefered taste regarding the CAPS and have been able to sort it out and altered it for myself... Possibly an option somewhere.

The Header/Footer is not a problem now. as I have added it in as option in the configuration menu and when producing new sheet it appears as I need it...

Attached is an example, it has a block for each letter... (I am creating wooden letters) and need to select each object separately to maximise cutting. )
The file attached shows an example of what I am trying to achieve. Again I have tinkered the code which now allows the option to separate each object when producing the text. I have attached a copy of the modified python file, be it a rough version at the moment but works well for me. Maybe an change for the future.

How do you refresh the code without reloading file. For example I manually add a line containing (Block-name: ####) but it does not add the block straight away, only when I reload the file. I have to do this to add a BLOCK and split an existing BLOCK i.e. separate the letters in the 'text tool'.
Or am I missing something, is there a way to split a BLOCK.

Tools 'Text.py' file changed to txt to attach it .
[text.py.txt](https://github.com/vlachoudis/bCNC/files/1428122/text.py.txt)

Example of modified output as I require it.
[Test Maisie - Copy.txt](https://github.com/vlachoudis/bCNC/files/1428099/Test.Maisie.-.Copy.txt)

closed 2018-10-29 by Harvie