vlachoudis/bCNC #689: setup bCNC to work with 2 machines

pietdep, 2017-10-30:
I was wondering if there is a way to set up bCNC with some kind of 'machine profiles' so I could use it on my CNC and my lasercutter (if possible simultaneously) ?
I'm running Bodhi Linux.
Any help is appreciated very much!
Piet

vlachoudis, 2017-10-30:
Yes, that feature is already there.
To start bCNC with a different configuration than the default "${HOME}/.bCNC" supply it
as a command line option like:
```bCNC -i myothermachine.ini```
the name/extension can be anything

pietdep, 2017-10-30:
Thank you for the swift response, I'll try that

pietdep, 2017-10-31:
I have been trying this approach but did not succeed
I copied bCNC.ini and renamed and adapted it with ePad (Xmax and Ymax) and then tried your suggestion but the program doesn't seem to want the adapted *.ini, it just goes with the original one
any suggestions?
thank you

vlachoudis, 2017-10-31:
the bCNC.ini is the master ini with the default values. It should not be touched. You should copy the ${HOME}/.bCNC   which contains only the user settings.

pietdep, 2017-10-31:
Strangely I dont have a ".bCNC" in my home folder ? Maybe because I unzipped it to my desktop folder ?

pietdep, 2017-10-31:
I removed the bCNC folder from the desktop and unzipped it in my home folder, still no .bCNC but I have 2 bCNC files (without the point), one being a shellscript, one a desktop-configuration file. Which one can I copy and adapt?

vlachoudis, 2017-10-31:
The .bCNC in the home is created and saved (at exit) automatically after you modify something inside bCNC

pietdep, 2017-10-31:
I succeeded to copy the file, rename ('bCNC-Laser') and adapt it (changed line 6 to 'Exec=bCNC -i bCNCLaser.ini' but still it takes the original ini.
I do have a .bCNCLaser.ini in my home folder.
Is it possible not being connected to the machine has an influence on the saving of the file. I'm setting up a new 'old' pc indoor and the machine is in the shed ...
An extra question, it would be nice to change the heading for the laser to'bCNC Laser' (top of the window), is that possible?

burtbick, 2017-11-03:
I just did a quick test with the latest version of bCNC downloaded 11/03/17, After running and making a change I have .bCNC in my home directory (show hidden files option for Linux).  I first tried copying the file to .bCNC2 and then bCNC2 when the first didn't work, so I suspect that might be your issue.  What I did that does work is:  1) run bCNC -i bCNC2.ini 2) Make a change and exit bCNC.  You should now see the bCNC2.ini file in your home directory.  Now if you start bCNC without the -i option it will load the original config file.  If you start bCNC -i bCNC2.ini it will load the new config file.  Of course name your config file(s) as you wish.  So easy way is to start bCNC -i new config file name, make your changes, exit bCNC and your changes should be saved to the new config file which you can load when needed.  Hope this helps.

pietdep, 2017-11-09:
Thank you for the response I'll try that approach
Piet

closed 2018-10-29 by Harvie