vlachoudis/bCNC #690: UnicodeEncodeError

caoxiangfish, 2017-11-02:
There is a error printed on the cmd when clicking the exit button:
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-1: ordinal not in range(128).

it's easy to solve this problem:
add the two line below in the beginning of bCNC.py after "improt sys":
`reload(sys)
sys.setdefaultencoding('utf-8')`

eg:

`import os
import re
import sys
reload(sys)
sys.setdefaultencoding('utf-8')
import pdb
import time
import getopt
import socket
import traceback
from datetime import datetime`

vlachoudis, 2017-11-08:
@caoxiangfish this is not recommended in python to do.

GaramMasala, 2017-12-19:
That's a workaround and it works. Now it's even possible to save files and load them. Will be running this until the problem is fixed. Thank you caoxiangfish.