vlachoudis/bCNC #692: G-code words consist of a letter and value. Letter was not found.

maikO84, 2017-11-09:
Hi I got an error while milling with autoleveling. I think this could be an Issue because the Gcode seems OK for me (There are Letters for XYC- see screenshot). I tried the same Gcode without autoleveling and had no failure.
[image]

CyberHugger, 2017-11-09:
The Z value in that command is really long. I wonder if something isn't prepared for that level of precision.

iceturf, 2018-06-10:
I noticed this issue myself, I found the only solution was to restart bCNC.

pb777, 2018-06-10:
Meantime i changed the USB-Cable and I never had this problem again, maybe it is Communicationfailure.

iceturf, 2018-06-10:
Curious - I have a CNC protoneer Hat, so I can't shorten or change my cabling. Good to hear you were able to clear up the problem.

Harvie, 2018-10-29:
Use latest version. If problem persists, try the editor tab -> edit -> "round" feature.

closed 2018-10-29 by Harvie