vlachoudis/bCNC #693: Spindle widget.

goloplo, 2017-11-15:
I have an IR sensor on Arduino that is counting RPM.
How can I introduce a widget/add-on on bcnc that will display current RPM of the spindle using Grbl and spindle?
Thank you.

chaotix-, 2017-11-16:
I don't think this is possible with a reasonable amount of work. Either you would need to modify GRBL to know about the actual RPM (I think there is one analog pin left that could theoretically used for something like this). Then you would need to modify the protocol GRBL is talking to also send out the RPM data over the serial connection and then you would need to modify bCNC to make sense of this data.
The other alternative would be to use a second serial connection to your Arduino which would send out the RPM information to the connected PC running bCNC. But then you would need to teach bCNC how to use a second serial port where a protocol that you would have to design is spoken to display the RPM somewhere in bCNC.

What about just adding a cheap OLED or LCD to your Arduino and just show the RPM there? This should be relatively painless to implement and is contained to your RPM Arduino.

goloplo, 2017-11-18:
Chaotic thank you for the answer.
I think I will try to develop what you have listed first. Because Uno is at the limit with GRBL my plan is to use Mega. I will try to replicate Probe script in GRBL with new pin alocated for IR sensor. I'm thinking to do the measuring at the GRBL level, and the to send them to a plugin on bCNC. The idea is there is a RPM before the tool hit the material and there is a lower RPM when the milling is done. With this the diference can be compensated. 
Please let me know if you have other idea of the implementation.

chaotix-, 2017-11-18:
If you just want the RPM data to adjust the actual spindle speed I would consider a different idea: Connect your Arduino between the GRBL spindle PWM output and the spindle analog input. Use the GRBL spindle PWM output as an input to your Arduino (with an RC filter to smooth out the PWM). You can then do the PID calculations in your Arduino and adjust the output voltage for the spindle accordingly. This should be relatively easy to implement and you would not have to modify GRBL, the communications protocol and bCNC.

goloplo, 2017-11-18:
But I still want to display measured RPM in program. I could be a good feature.

Harvie, 2018-10-29:
This is out of bCNC's scope. There are request to add RPM feedback loop to GRBL. I believe this will solve your issue.

closed 2018-10-29 by Harvie