vlachoudis/bCNC #694: Pendant flickering

Guenni75, 2017-11-17:
Hi.
I do not often use the Pendant.
Yesterday i opened it on my Phone and the jog button were jumping up and down.
There is an image that could not be loaded (i think).

The Windows Python window has an error:

----------------------------------------
Exception happened during processing of request from ('192.168.178.51', 50662)
Traceback (most recent call last):
  File "C:\Python27\lib\SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Python27\lib\SocketServer.py", line 655, in __init__
    self.handle()
  File "C:\Python27\lib\BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "C:\Python27\lib\BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "D:\bCNC\bCNC-master\Pendant.py", line 124, in do_GET
    Image.open(tmp.name).save(out.name, 'GIF')
  File "C:\Python27\lib\site-packages\PIL\Image.py", line 1952, in open
    fp = __builtin__.open(fp, "rb")
IOError: [Errno 13] Permission denied: 'c:\\users\\cnc\\appdata\\local\\temp\\tmpm4tqld.ps'
----------------------------------------

I think Permission denied is the Problem, but "run as Admin" could not fix it.

Or is it the double \\? But i think this is only from the error handler.

If i delete the image on the html site with chrome, all is ok.

Effer, 2017-11-17:
Confirmed on WIn10.
It maybe be related to different way OS handle temporary file:
https://stackoverflow.com/questions/23212435/permission-denied-to-write-to-my-temporary-file

vlachoudis, 2017-11-18:
According the tempfile documentation it opens a file in the default directory. Have you tried setting the TMP or TMPDIR, TMP or TEMP system environment variable, as well to have a user writable C:\TEMP or C:\TMP directory?
https://docs.python.org/2/library/tempfile.html#tempfile.tempdir

Effer, 2017-11-18:
```python
with tempfile.NamedTemporaryFile(suffix='.ps') as tmp:
	httpd.app.canvas.postscript(
		file=tmp.name,
		colormode='color',
	)
	tmp.flush()
	with tempfile.NamedTemporaryFile(suffix='.gif') as out:
		Image.open(tmp.name).save(out.name, 'GIF')  #<< Trying to open the already open tmp file???
		out.flush()
		out.seek(0)

		self.do_HEAD(200, content="image/gif")
		self.wfile.write(out.read())
```

From docs:
> Whether the name can be used to open the file a second time, while the named temporary file is still open, varies across platforms (it can be so used on Unix; it cannot on Windows NT or later).

In Windows you cannon't open again a file returned by NamedTemporaryFile(). But it seems exactly what the code is trying to do in Image.open.
I don't think its a matter of privileges.

vlachoudis, 2017-11-18:
Thanks @Effer the file is already open and locked.

Guenni75, 2017-11-18:
At every "blink" the filename changes ('c:\users\cnc\appdata\local\temp\**tmpm4tqld.ps**').

Guenni75, 2017-11-23:
When i look in the temp folder at the end of the message :'c:\users\cnc\appdata\local\temp\tmpm4tqld.ps' there is no file with the ending .ps .
Is there a problem creating the file?

Guenni75, 2017-11-26:
The Problem ist the NamedTemporaryFile().
Mentioned on [docs.python.org](https://docs.python.org/2.7/library/tempfile.html), this could **not** be used under Windows (since Windows NT)!!

vlachoudis, 2018-01-29:
I could not find a reasonable solution. So for the moment in case of error, it shows a warn.gif icon to stop the flickering.

closed 2018-01-29 by vlachoudis