vlachoudis/bCNC #695: USB numpad

goloplo, 2017-11-18:
I know that the usb keypad is already mapped by default, but is there any method to use the shortcuts to custom map this type of numpad?
[image]

vlachoudis, 2017-11-29:
Unfortunately not through any configuration. It needs some tweaking in the code.

closed 2017-11-29 by vlachoudis

einencool, 2018-04-29:
Hello,

Would it be possible to add the macros to the Shortcuts?
How much work would it be? I don‘t know much about programming, but it would be very nice to add the macros to the shortcuts, so I could map them to the wireless Num Pad and it would make controlling the CNC much easier.

Thank you in advance.