vlachoudis/bCNC #696: Error when closing

GaramMasala, 2017-11-20:
Program     : bCNC
Version     : 0.9.11
Last Change : 2 Oct 2017
Platform    : freebsd10
Python      : 2.7.13 (default, Aug 15 2017, 20:21:43) 
[GCC 4.2.1 Compatible FreeBSD Clang 3.4.1 (tags/RELEASE_34/dot1-final 208032)]
TkVersion   : 8.6
TclVersion  : 8.6

Traceback:
Traceback (most recent call last):
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/Utils.py", line 398, in __call__
    return self.func(*args)
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/tkExtra.py", line 2719, in fileDialog
    filetypes=self.filetypes)
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/bFileDialog.py", line 925, in askopenfilename
    return OpenDialog(**options).show()
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/bFileDialog.py", line 366, in show
    self.typeCombo.set(self.typeCombo.get(0))	# will fill the files
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/tkExtra.py", line 3061, in set
    self.invoke()
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/tkExtra.py", line 3013, in invoke
    self.command()
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/bFileDialog.py", line 765, in typeChange
    self.fill()
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/bFileDialog.py", line 595, in fill
    self.fileList.sort(0, False)		# First short by name
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/tkExtra.py", line 1800, in sort
    MultiListbox.sort(self, column, dir)
  File "/usr/home/bernt/nerladdningar/vebb/bCNC-master/lib/tkExtra.py", line 1730, in sort
    elements.sort(key=self.sortAssist(column), reverse=reverse)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe5 in position 3: ordinal not in range(128)

GaramMasala, 2017-12-17:
We do not use unicode/International characters in the filenames. One can't save a file with this filename polygon_areas.ngc and open it you need to specify it on the commandline.
Does not show up in the file chooser.

GaramMasala, 2018-01-01:
There is no problem with pycam or flatcam. The problem is in bcnc, not python.

onekk, 2018-01-17:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe5 in position 3: ordinal not in range(128)
This is a python error regarding utf-8.

Python2 will manage utf-8 but some functions don't correctly decode the utf and so this error is raisen.

It seems to me that probably is the file-chooser part that raise this error and probably the error is raisen when in the directory you search in there is a filename with unicode characters in it.

I have had some problems sometimes and I've resolved to use a proper directory (dedicated) where i put the files for machining with bCNC, with proper names.

just my two cents.

Regards

Carlo D.

Harvie, 2018-10-29:
this will get fixed with python3 support

closed 2018-10-29 by Harvie