vlachoudis/bCNC #698: Tool Change Calibration/Probe causes alarm

berkakinci, 2017-11-26:
Hi @vlachoudis 
Great job on bCNC.  I'm finding it very useful and elegant.
I just started using tool changes and ran into a couple of problems.
1) manual tool change with no probe was not useful because I could not manually reset the WCS at the time.  I had to work around by stopping and disabling blocks of code.  Good news is, it got me down the better path of attaching the probe...
2) When trying to calibrate (possibly related to issue #671) I got alarms because the generated probe commands set a destination outside my machine limits.
I'm running 0.9.11  (despite About reporting 0.9.10)
The code responsible still seems to be the same on master:
https://github.com/vlachoudis/bCNC/blob/8636836c54428df73892d4e54233af5af9ec36a4/ProbePage.py#L1691

Symptom on terminal:
`Grbl 1.1e ['$' for help]`
`g53g0z0.0
ok
g53g0x0.0y0.0
ok
g53g0x0.0y0.0
ok
g53g0z-50.0
ok
g91G38.2f180.000000z-34.9
[PRB:0.000,0.000,-82.633:1]
ok
g91G38.4f180.000000z34.9
[PRB:0.000,0.000,-82.610:1]
ok
g91G38.2f60.0z-34.9
ALARM:2
[MSG:Reset to continue]`
That last probe command goes outside my machine bounds.  I'm not sure what the best policy for all users would be.

berkakinci, 2017-11-26:
User workaround:
Disable fast probe feed by setting "Common -> Fast Probe Feed" to 0.0
[image]
(BTW, units are not quoted.  They happen to be GRBL native feed units; mm/min in my case.)

I'm pretty convinced now this is a duplicate of #671.
I'll leave it open for your review anyway.

vlachoudis, 2017-11-28:
Could you try with the new master. Now it tries to retract back to the toolprobez position, so it wont be outside the machine coordinates

goddisignz, 2018-01-22:
Hi,
I am using the current master and had the same problem. Although the z-coordinate for the reverse probe is set to go back to the toolprobez position it stops after it looses the contact to the touchplate.
My current solution is to simply retract by 1mm (never reached) and probe again by 1mm.
Maybe using absolute MCS coordinates instead of relative positioning is a better way to avoid exceeding the limits.
I opened a pull request but just found a mistake so I closed it.
I will reopen it when I found a correct behaviour.

vlachoudis, 2018-01-29:
@goddisignz this is what I've tried to do in the new version. I cannot use machine coordinates for the probing, but the code is converting it to work coordinated. Maybe I mess it somewhere.

Harvie, 2018-10-29:
@goddisignz @berkakinci Does that solved your issue?