vlachoudis/bCNC #699: Drill Macro Expansion: G98 and G99 sent to GRBL

berkakinci, 2017-11-26:
G98 and G99 commands seem to be handled by bCNC.  (it sets retractz variable.)
However G98 (and presumably G99) are also sent to GRBL.  This causes an error returned:
`G90
ok
G98
error:20`

These should be consumed by bCNC for Drill macro expansion but not sent to GRBL.
I spent a bunch of time trying to figure out where that would fit, but I couldn't figure it out.  (shame...)
Where would the removal of commands be done?

vlachoudis, 2017-11-27:
You are right, they should be skipped. You can avoid sending them to grbl by setting
with a text editor the following variables in the user ${HOME}/.bCNC   initialization file, in the [Error] section
```
[Error]
G98=1
G99=1
```

berkakinci, 2017-11-28:
Thank you.  I've added to my file (then re-added after I actually closed the application. :-) )
I will get a chance to try it once I'm back at the machine.

vlachoudis, 2017-11-28:
I've added it inside the code when the drill policy is to expand

Harvie, 2018-10-29:
I understood this is fixed. Please let me know if i am wrong.

closed 2018-10-29 by Harvie

be-engineer, 2022-05-21:
Can I delete directly this G99,G81 code from .nc file?

berkakinci, 2022-05-21:
Has the bug returned, or are you asking in general?  (I'm struggling to remember this old bug.)
If you're asking specifically about bCNC, or anything else that expands drill cycles: no, if you delete these you will be missing retract setting and the drill action.  bCNC should be able to expand these drill cycles into other G-codes.  Alternatively, your CAM post-processor can also expand them.

Ericounet, 2024-08-10:
hi,

I'm testing the Cam workbench with Grbl gcode export ; the G98 and other G90 etc. are still present and cannot be processed par grbl.

A good idea would be to give the equivalent gcode for every gcode command not usable under Grbl.

So we could add it to our CAM-postprocessor.

Thanks for the hard work.

Eric