vlachoudis/bCNC #700: geometric error compensation

Ravenbs, 2017-11-28:
Feature request:
All major vendors of CNC machines offer a software compensation for geometric machine errors.

Most important is to calibrate all three axis direction.
So for example the direction of x axis could be 0.99x and 0,01y

Minor compensations are offsets, depended on the position of the other axis.
As example the so called slack error.
Depending on the position of the Y axis the portal of a big machine could bend and the z position could compensate this.

Ravenbs, 2017-11-29:
Discussed also in GRBL.
The think it is a GUI problem to solve?

https://github.com/gnea/grbl/issues/313

vlachoudis, 2017-11-29:
It could be an interesting feature. There is already something, which maybe with a bit of tweaking can do what you want. If you go to the Tab "Probe -> Probe" There is a panel called "Orient"
This is used to orient a gcode based on a list of markers correspondence of gcode coordinates to machine coordinates. For the moment it only applying rotation and translation, however the matrix if it is relaxed the condition of orthogonality it can handle any skew one.
The only drawback is that in case of skewed transformation the arcs, g2,g3 will be converted to ellipses, and therefore should be expanded to linear segments, as it is done in the autoleveling.

Of course instead of re-inventing the wheel, since I am completely ignorant on machining, if you can provide some more information on how the compensation info has to be given to the program, it will be very helpful.

Ravenbs, 2017-11-29:
To measure the machine, I use a dial indicator and a set square (hope this translated right?)
First I align the set square to Axis X.
For example by milling a straight pice of wood as dolly.
Then I use the dial indicator to measure the Y deviation in some distances.

After this small process, I know to have 0,123 mm deviation in 400mm distance.

So the input could be degrees or deviation in specific distance. 
I think most important is this x/y squaring over the other axis.

Another method:
Using the probing. Set the set square under current and probe it 2 times on x and 2 times on y...
Should give the same values as above, maybe easier.

flounderscore, 2017-12-12:
> The only drawback is that in case of skewed transformation the arcs, g2,g3 will be converted to ellipses, and therefore should be expanded to linear segments, as it is done in the autoleveling.

@vlachoudis Wouldn't it make more sense to approximate the ellipse with multiple (circular) arc to increase the precision?

Ravenbs, 2018-01-15:
After a lot of research, I found that this squaring problem is very common on machines with slave axis.
Normally it is solved by using an expensive controller like Beamicon2 or Eding (400-500€).

But I found a way to do this easily in BCNC.
Find attached my code for BCNC. 
[autoSquaring.txt](https://github.com/vlachoudis/bCNC/files/1631619/autoSquaring.txt)
Please consider including it in further versions, it is only a single file modified.

1. Place a set square on the table and connect probing
2. Place your endmill in the inner corner of the set square (inner corner ist in x+ and y+ direction)
3. Run the Code:
- It touches 4 points on the set square
- It calculates the current angel and also the offset to compensate this angle and popps up a dialog.
4. Disable one of your stepper motors (for example with a switch in it's enable wire to the motor driver or by unplugging it). It must be the motor on the "-" direction of x or y. Same code works for both axis.
5. Klick "ok" in the bCNC popup, and it will move the remaining motor for the compensation amount.
6. Enable your second motor again.

-> Now your axis are perfectly aligned in only 20 seconds.
Without the need for a second and very well calibrated endstop or an expensive 4 axis controller.

souravg009, 2018-03-10:
Hi,
I'd like to add an idea for "CNC Squaring". I am an Electrical Engineer and I'm not so good with codes, thus I can only suggest.
Idea is to make a predefined template of markers on the CNC bed itself; like the "prusa i3 MK2" 3d printers can auto-calibrate its all 3 axis. in the same manner, a CNC should be capable of self-calibrate its orientation in the "Machine profile / ID" settings. this settings will be applied for all G-codes operating on the selected machine. For another CNC settings may be different.
Also apply this system for both probe and camera squaring. I think it should do.
- Sourav Ghosh.

Ravenbs, 2018-03-11:
Thank you for your feedback.
What is Camera Squaring?

My CNC ist at least 10 times more accurate as my 3d Printer.
Where I try to reach 0,1mm accuracy on my Printer and fail badly I try (and can) reach 0,01mm on the cnc.

As far as I know, the Prusa uses magnets below the build plate with an hall sensor.
Unfortunately they are very strong affected by temperature and a few degree change in ambient temperature will give a different calibration.
And: Where to geht this very percise angle out of magnets? 

So the idea was, to buy a cheap but very percise set square according to DIN 875/II.
For 100€ you get a calibration within a few 0,001mm (1um).

souravg009, 2018-03-12:
Thanks for your comment Ravenbs. 
By the term "camera squaring" I intend to mean the method of squaring XY axis by a template (for example 'very precise set square' as you mentioned) or a very accurate drawn / printed template on **CNC bed itself** with a few "Marker holes" ( @instead of magnets as in heat bed in prusa i3).
In bCNC, we can feed the coordinates and calibrate those "Marker holes" of known / user defined coordinate with "gantry mounted camera". Similar to the calibration operation of prusa i3, but more accurate.
Thus we would be able to calibrate any machine's XY axis skew with our bCNC software. This skew settings can be saved for a particular cnc (in case of having multiple CNC). bCNC has to perform this transformation for all G-codes running on that CNC. after this transformation "ORIENT" operation can take place.
I have searching internet for quite a long time and i found another guy did something similar to "ORIENT" with " Affine transformation" to translate the coordinates.  Affine transformations can also be used for rectification of "Skew" error.
I hope those information may be of your help.
I am from India, please don't mind my bad English. also I don't Intend to be rude, but I have build my CNC within INR 6000/- which is almost 75€.  :)
- Sourav Ghosh

Harvie, 2021-01-07:
To get backlash compensation please consider using grblHAL instead of GRBL. It will probably not run on atmega, but upgrade is cheap (less than $10). grblHAL can be compiled with backlash compensation support (as well as other improvements to original GRBL). It is now officialy supported by bCNC.

https://github.com/terjeio/grblHAL/

Ravenbs, 2021-01-08:
Thank you for the tip, looks very promising! 
But I Switched to ESS in the meantime and are happy with it.