vlachoudis/bCNC #703: Use bcnc as remote interface for another instance of bcnc over network

makenb, 2017-12-04:
First of all thanks for this great tool! I use it to control a paper cutting machine I made by modifying a cricut machine to run grbl. I have an idea for an enhancement, I don't know how difficult it would be or if even possible. I think it would be very useful to be able to control an instance of bcnc remotely with bcnc, kind of similar to the http remote. Maybe you could use the http.interface or maybe it could be more like chilipepper with a virtual serial port. Thanks for listening,
Makenb

MrRemmers, 2018-01-02:
Have you looked at the pendant feature?  The pendant is a web interface that allows remote control.

makenb, 2018-01-02:
@MrRemmers 
Yes of course, The pendant is actually what gave me the idea. the pendant seems to have all (most) of the relevant information to be able to update the main interface (eg. coordinate info could be used to update the tool location in the gcode visualizer), maybe not quite as real time but still useful. but it would be very convenient to be able to use the gcode editor or the tools menu and then immediately upload the file to a remote machine and watch the progress as if it were local, seamlessly.

MrRemmers, 2018-01-02:
@makenb 
I think I understand now.  Have you happened to see this: #258  ? I haven't been able to use the pendant because of flickering described in #694, so i'm too ignorant to know if you can open up a gcode file from the pendant (either on the local or remote machine).  But if that's possible is it helpful to propose editing with bCNC on the remote machine, saving the gcode, then opening that file on the machine connected to the paper cutter and running the file.
Are you familiar with Octoprint?  that is kind of the toolchain I'm describing.  make the g-code on a PC then upload and run from the octoprint server connected to the 3D printer.

Harvie, 2018-10-29:
There is software available that allows you to access serial port on remote machines over IP.