vlachoudis/bCNC #708: Z axis precision problem

Thorgrlm, 2017-12-20:
Hi everyone, I have a problem with the Z axis. With the same Gcode, it never cuts at the same height. this is happens only in axis Z

I want to make PCB. 
My CNC is a 3040, with Arduino Uno + Shield + GRBL + TB6600 and I use the bCNC software

I do the autoleveller process, the copper thickness of the PCB is 0.035, I make a Gcode that cuts to -0.037, I do the work and I wait at it ends.
Without changing anything, neither tool, nor PCB, with another Gcode of screen printing at the same depth of cut and using the same values of autoleveller, I start the CNC and cut well below -0.035. sometimes the PCB has been destroyed 

I installed a gearbox of 1:10 and continue with the problem

I have created a Gcode whit very long time repeating movement only from the Z axis (for example, moving 200 times from 10cm to -10cm) when it finish, I measure if it have lost a step and don't lose steps.

Do you know in any way to discover what the problem is?

Thanks u

Guenni75, 2018-01-09:
Hello.
I have the same Problem anytime i use autolevel on big jobs.
Only dividing in smaller blocks and probing between them really helped.

Running the big job without autolevel works perfect.
I make 3D Milling (very much Z-moving!) too and no problems with the Z-Axis.

Thorgrlm, 2018-01-09:
Hi, I have solved the problem, in my case I had not tightened well the screws fastening the motor with the Z axis.

What model of machin you use?

closed 2018-01-29 by vlachoudis