vlachoudis/bCNC #710: bCNC / GRBL Freezing Up

MikeMTM, 2017-12-31:
Hi everyone. I have been experiencing and issue for quite some time now and I'm really not sure what the root of the problem is. I don't always get an error message but the my machine will freeze up at the start of a job once or twice a day. This is usually only after executing a couple lines of code on a really simple operation. Once the machine freezes up, the only way to get things moving again is to do a soft reset of bCNC. I've been just dealing with it but it's actually somewhat dangerous because the spindle stays on also until after the reset. When I close down bCNC I get a small error window that opens up and appears to be designed to automatically send to someone but it never completes even after entering my own email. A couple weeks ago I started using the windows snipping tool to capture the error messages. I have a few more that I saved but this should do for now. I think a couple of them state the same thing but they are all from different times. 

Thank you to everyone who is willing to help me.

[image]
[image]
[image]
[image]
[image]

MikeMTM, 2017-12-31:
I'm trying to if maybe there is some sort of system update that is needed. My biggest issue is this computer is sort of my dedicated junker for the cnc so I tend to neglect the system updates.

GnomeTek, 2018-01-12:
I also have been getting "sender.py" error logs at close of bCNC. my system has frozen several times also. 
win 10, basic computer purpose for running cnc alone. grbl 1.1

MikeMTM, 2018-01-21:
So nobody has any idea what's going on?

MikeMTM, 2018-01-21:
Often times after a freeze & restart, I get the "error 35"

GnomeTek, 2018-01-22:
I've got a Shapeoko 3 running the Carbide Motion Control board from mid year 2017. i believe the V2.4 PCB. Shapeoko's wiki is mixed with new and old information so it's hard to get a great reference for this kind of stuff. On top of that they do not have a central information depot for their controller revisions or schematics. Although I did find out that the SparkFun Stepoko board is the same schematic/BOM as the Carbide board at least at _some_ version lol. So since I'm not at my machine to check part #'s right now I'll go with that. 

USB interface is done via FT232RL to Atmega 328 for processing. running GRBL V1.1 firmware w/ no mods - that is to say, the firmware was direct from Shapeoko for their machines. So it may not match bit by bit with GRBL V1.1 via GitHub.

MikeMTM, 2018-01-22:
I spoke with the designer of my board and he's fairly confident that this is software related. He recommended I uninstall bcnc and any drivers I have installed and start fresh. I hope I can get this to work because I prefer bcnc over anything else.

MikeMTM, 2018-01-22:
It seems like once the freeze occurs, I can restart and continue without any problems.

timowen01, 2018-03-06:
I've had the same hangup issues with bCNC & the smoothieboard.   It doesn't seem to finish a simple g code file ( or sometimes during manual operations) without saying ALARM Abort during cycle.  I can sometimes clear it by going to 'file' and disconnecting and reconnecting the serial button - the message disappears and sometimes but not always things go back to normal. If not I have to physically disconnect the USB port, or occasionally restart bcnc.  I am sure its in bCNC!

vlachoudis, 2018-03-06:
Can you send the gcode file to check.

MikeMTM, 2018-03-06:
For me, the freezing happens once every day, no matter what gcode file I run. It's a guaranteed occurrence. After bCNC freezes, I disconnect from the board then reconnect. I'm usually ok after this but I have to be careful because the machine will run a few lines of the code before freezing so it can destroy my part. My work around is to set my zero away from the part and mill in the air until it freezes. After it's done the "daily freezup", I can re zero on my part and things go ok.

timowen01, 2018-03-06:
Thanks for getting back so promptly - its a great piece of software if 
only!.....
It happens for any code and sometimes just for manual moves too.
When I run code (only small, simple test code so far a couple of 
different ones of 6-12 lines or so - see attached) it goes to the end 
and stops ( says 93% run) but the  run arrow on the control tab doesn't 
come on and the motors are still energised- locked. If I then press the 
stop on the control tab  the run arrow lights and the motors depower.   
I have then lost the link to the smoothieboard - resetting the 
smooothieboard (its own reset button) doesn't get the COM link back. 
Hitting the bCNC reset button top left doesn't get it back ( that is 
just resetting the smoothiboard?)  To get the link back I have to go to 
the FILE tab and click CLOSE and then OPEN on the coloured COM box ( or 
type CLOSE then OPEN in the command line).  That almost always gets it 
back but sometimes I have to remove the USB cable as I get a box saying 
it can't find the COM port.  On occasions I seem to have to kill bCNC 
and reopen it.
If I press the reset key top left on the screen when it is 'dead' after 
a manual move,  I get a message ALARM-Abort in .... in red. The red box 
otherwise always shows 'NOT CONNECTED' even when its working. The 
terminal text shows this message when the program stop button in CONTROL 
is pressed after it has run.  I have never got it to run two programs 
one after the other without closing and opening the COM.  I don't think 
there is an 'end run' g code I've missed?
I'm running Windows 10 on an ASUS laptop.

Tim Owen
www.cablesfarm.co.uk
1 Station Road
Dullingham
Newmarket
Suffolk  CB8 9UP
01638 507192 / 07768 757056

G17 G21 G90 G94 G54
G0 Z0.25
G01 Z0.
G01 X0. Y30.
G01 X0.5 Y30.
G01 X.5 Y0.
G01 X1  Y0.
G01 X1. Y30.
G01 X1.5 Y30.
G01 X1.5 Y0
G00 X0. Y0. Z0.25
M30 RG
G

 
G17 G20 G90 G94 G54
G0 Z0.25
X-0.5 Y0.
Z0.1
G01 Z0. F5.
G02 X0. Y0.5 I0.5 J0. F2.5
X0.5 Y0. I0. J-0.5
X0. Y-0.5 I-0.5 J0.
X-0.5 Y0. I0. J0.5
G01 Z0.1 F5.
G00 X0. Y0. Z0.25

colinluthier, 2018-03-07:
I had lots of problems like this before I replaced my serial cable with one that had a ferrite core. Also, I set my switches to normally closed.

vlachoudis, 2018-03-07:
If it happens with any code even on moves, then it is noise problem. Check your cables, put capacitors on the switches, ferrite, etc... There is extensive information on that on the web.

timowen01, 2018-03-07:
Thanks, I'll do that, but its really too repeatable to be noise - I've 
been trying it today with a longer g code.
It NEVER drops out in the middle of the code.  When the code has 
completed all the steps all the CONTROL functions etc greys out (except 
pause and stop)  so you can't tell anything but it is ALWAYS drops out  
when I press STOP, at which point the terminal window says in red;-   
ALARM- Abort during cycle - lfcr -  Run ended - lfcr - date and time, 
then e.g. Current 15 [15] completed 93%.
It now seems to allow me ALWAYS to do manual moves without dropping the 
comms.

If you run a g code bit and it finished, and you leave the CONTROL tab 
alone ( greyed out) and CLOSE & OPEN the comms on the FILE tab, the 
CONTROL tab becomes active but hitting a manual move gets the error 
message in red on terminal ; error Alarm lock, and a g90 and a $G and 
the report but the move doesn't work  - this suggests to me that there 
is still comms even when you can't get any response from the axes. If 
you CLOSE and OPEN the comms a SECOND time the move buttons work.   This 
is really strange!

One possibility is that STOP releases all the motors at once and that 
might cause a spike.  But I don't get any problems from turning the 24 
volt supply on and off. Or when it hits an endstop and stops with all 
the motors suddenly released - the comms don't  seem to go until you 
send a RESET, when the red status  ALARM display message pops up. The 
red display status always says NOT CONNECTED except as above when it 
says ALARM etc.  It never says CONNECTED.  The terminal display ALARM 
message ALWAYS accompanies the loss of COMMS.

   But that doesn't explain why it never thinks it has completed a run 
on its own  and only thinks it has run 93% or thereabouts and gives the 
error message when you press STOP even though it has run every g code.

How is a g code run supposed to end?  Does it normally leave the motors 
energised?  Should it wait for the STOP to be pressed?
Is there a g code for releasing the motors, I can't find one?
What triggers the red ALARM notices? - the big red button doesn't 
usually show the message, but its always on the terminal whenever the 
comms goes.

I'll have another go with the noise precautions, but the high quality 
ferrited usb lead I tried doesn't make any difference.  I'm beginning to 
think its maybe a bug in communication between the smoothieboard and 
bCNC.  I probably need to monitor all the comms between the two if I 
want to track it down...........

Thanks for your help.

Tim

Tim Owen
www.cablesfarm.co.uk
1 Station Road
Dullingham
Newmarket
Suffolk  CB8 9UP
01638 507192 / 07768 757056

colinluthier, 2018-03-07:
The classic test for noise is to make it glitch with the spindle off. If it still glitches spindle/router off its not noise.

vlachoudis, 2018-03-07:
@timowen01 I can always make a test run on my machine if you can give me the gcode. The SerialExceptions can only happen if the serial line drops while bCNC tries to write to it
 bCNC is never closing the serial connection except in the case of "Hard Reset", which can only be triggered by the user. Therefore if I see SerialException for me it points that something happens and the serial connection is dropped. It can be the OS, driver or noise.

Harvie, 2018-10-29:
Is this still a problem?

Awetomation, 2018-10-30:
I changed to a windows 10 machine and still have the same problems but not as bad as the Linux machine. I did test with the spindle off and with the store drives off as well on my Linux machine and the same thing happened. As stated the windows machine is not as bad but it will always hang when I press stop or pause and I have to re-establish communication again.

pda3k, 2019-01-07:
I experienced the same problem mentioned in this thread as well as a similar one.  I am running a new Macbook Pro with the latest OS, Python 2.7, and latest bcnc version (0.9.14) against a Smoothie board that has been flashed with the latest stable update.

1.  Freezing up in the first few lines of gcode being run.

I ended up going into the "Tools" tab and selecting "Config".  I deleted the header and footer gcode that is there by default.  Interestingly, they both look like a single line until you select them.  They then show up as two lines.  I'm not sure if that has anything to do with the problem but after I deleted these two settings, my freezing up problem in the beginning of a run went away.

2.  Freezing up after running gcode and jogging around.

This problem took a bit of poking around and debugging to somewhat figure out what might be the issue.  On my system, it appears that the sender.py - def serialIO(self): function sline and cline lists don't remove entries if you manually enter commands (like jogging).  Once these lists get to a certain point (count?), commands stop executing.  I may have fixed it but I am not sure exactly how all of this code works together and I know threaded routines can be very tricky to figure out.  What I did seems to work and not impact any other areas and I have tried this on several large files and lots of jogging around.  I worked all afternoon with no freeze ups or issues after making this change.  I am sure there is probably a better way to do this but someone smarter than I am might get some clues from all of this.  If anyone else experienced this and has a better fix, please let me know.  As of now, all is working well.  The original code section and modified code is below.  This code is at the very end of the sender.py file.

NOTE: Tabs seem to not show up in this forum (at least in preview mode) but they are in the code

Original:
tosend = None
if not self.running and t-tg > G_POLL:
	tosend = b"$G\n"
	sline.append(tosend)
	cline.append(len(tosend))
	tg = t

Modified:
tosend = None
if not self.running and t-tg > G_POLL:
	tosend = b"$G\n"
	sline.append(tosend)
	cline.append(len(tosend))
	# delete the contents of both sline and cline lists - also clear the buffer display
        del sline[:]
	del cline[:]
	self.log.put((Sender.MSG_CLEAR, ""))

Harvie, 2019-01-07:
The orifinal issue is reported for bCNC 0.9.10, which i would call very old. Can you try latest git bCNC with grbl 1.1f?

pda3k, 2019-01-08:
This was version 0.9.14.  Is there a newer version I should be using?

Harvie, 2019-01-08:
@pda3k yes. there was 0.9.15 release long time ago. but even that is old by now. you should try with latest git version. Or at least pypi version (which is right now almost synced to git version):

https://github.com/vlachoudis/bCNC/wiki/Installation

pda3k, 2019-01-08:
Thanks - Wow am I out of date!  I will give it a try.

Just out of curiosity, I used the "Clone or Download" button on https://github.com/vlachoudis/bCNC.  Is this not the latest version?
[https://avatars0.githubusercontent.com/u/3834594?s=400&v=4]<https://github.com/vlachoudis/bCNC>

vlachoudis/bCNC: GRBL CNC command sender, autoleveler and g-code editor - The world's leading software development platform · GitHub<https://github.com/vlachoudis/bCNC>
bCNC. GRBL CNC command sender, autoleveler, g-code editor, digitizer, CAM and swiss army knife for all your CNC needs. An advanced fully featured g-code sender for GRBL. bCNC is a cross platform program (Windows, Linux, Mac) written in python.
github.com

https://github.com/vlachoudis/bCNC/wiki/Installation

Harvie, 2019-01-08:
> I used the "Clone or Download" button on https://github.com/vlachoudis/bCNC. Is this not the latest version?

It is the latest version, but it is not 0.9.14 :-) Lot of stuff was changed, but there was no 0.9.15 release, so we didn't changed the version number displayed in UI... we should probably put something like "devel" in there... I will make a note about this...

pitass82, 2022-12-23:
Install St Link solved this problem of mine - Sw/get latest from https://www.st.com/en/development-tools/stsw-link007.html. Can move gantry without interruption any more.

Harvie, 2022-12-23:
> Install St Link

What? Why?

pitass82, 2022-12-26:
if you have issue in grbl / bcnc with interrupting serial com f e manual move w gantry

vadimyar, 2024-06-26:
[SOLUTION (possibly...)](https://github.com/vlachoudis/bCNC/issues/1808#issuecomment-2100233973)