vlachoudis/bCNC #713: GRBL Ver 1.1 works fine?

marceloabranco, 2018-01-03:
I'm wondering if version 1.1 of GRBL is working fine in bCNC. I would love to use some features that 1.1 brings, like instant change feed rate. 
Thanks

gibo77, 2018-01-03:
I have grbl 1.1, but could not change simple parameters on the fly with bcnc.

atrueresistance, 2018-01-04:
I've been using grbl 1.1 for quite some time without any issues, although I can't say I've tried adjusting parameters.

GnomeTek, 2018-01-12:
I have started on my bCNC adventure on GRBL 1.1 and i was not able to change parameters via check box. I could send  a command via command line and that seemed to stick. 

So far I'm unimpressed, frustrated, and coming to the project page to see if people really are able to use bCNC with any sort of sanity.

wwjohb, 2018-01-17:
Hey I have a GRBL 1.1 and bCNC setup, what specific feature is in question? the Feed Rate only thing tested? What OS platforms are you running bCNC on? I have noticed some differences on case-sensitive OS's

gibo77, 2018-01-17:
Same thing with me. I use GRBL v1.1 and the only way I can change settings is the command prompt at the bottom. But other than that, so far I was able to mill PCB just fine. Also I could not change parameters on the fly, as touted in GRBL v1.1 .

Guenni75, 2018-01-18:
I had BCNC on Win10 and everything works fine.
After changing to Linux i have the same Problems with changing settings with the check boxes.
You could not do anything, since I press STOP.

RobCNC, 2018-02-07:
I run bCNC on the ASUS Tinkerboard on TinkerOS 2.0.4. After Connecting to GRBL you need to press UNLOCK. The Status change to IDLE. Then you can enter new values on the Controller parameters. Once you hit the Controller button above the parameter table, the new values are sent to GRBL.

jpascher, 2018-03-27:
At the moment i looks like one has to use edit userfile Tools->pulldown - config ->edit userfile.
don't save before you close bCNC.  Save edited file and then start bCNC again. If on did try to use the button Controller then bCNC is blockt to get go to the other tub Control and push stop. Then restart COM connection within pulldown File push close /  open. I use windows 10 and grbl 1.1 hope this will be fixed soon.

neomadic, 2018-05-19:
I'm having no luck with grbl 1.1, bcnc, running on a raspberry pi-3.  Everything looked good during setup and testing but now it locks up randomly, flashes strange errors, and quits running code after a few cuts.  Researching how to debug now.  I hate to say it, but I'm really missing Mach3's diag screen!