vlachoudis/bCNC #716: PWM Freq.

DavidA50, 2018-01-07:
I want to convert the 5v PWM into a 0-10v signal to set the speed in a VFD. The issue seems to be the converters  available  will accept a PWM with a freq. between 1 and 3 KHz.  In looking at available data it seems the current PWM freq. in GRBL is between .98KHZ and  2MHZ. I was able to open and look at the cpu_map.h file and it seems to show 4 different prescaler settings.  

Knowing this does me no good as I do not know what or how to alter these settings and then get them to run on my system. Any help would be appreciated.

closed 2018-01-07 by DavidA50

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68

MARIOBASZ, 2020-11-23:
Read the manual for your VFD:
I control the speed of mine with pwm 0 - 5V

LYHTSPD, 2020-11-23:
Mario,
That is the same way I am controlling mine.  Seems to work really well so
far.

-Brandon Dilger