vlachoudis/bCNC #717: Spindle Start then stop when bCNC loads

Biggs427, 2018-01-08:
I'm running bCNC  latest version under linux on Raspberry PI.  Everything works fine except the when I launch bCNC the spindle starts automaticly runs for 1-2 sec then stop.

Once loaded, the start/stop and RPM control are working flawlessly.

What would cause such behaviour?

Thanks

tuxintrouble, 2018-01-09:
As far as I know this is a GRBL/Arduino problem. Ehen bCNC connects, it resets GRBL which toggles the Arduino's Spindle output.I have routed my pwm wire through the e-stop switch, which I have locked when I start bCNC. Otherwise there is not much you can do about it IMO.

lalo-uy, 2018-01-10:
You can remove de 100nF cap ( or the R in the V4) to avoid auto reset at
port openig. You will have to manual press RESET to flash after that.

closed 2018-01-29 by vlachoudis