vlachoudis/bCNC #720: ‘MPos’

Svetkobuha, 2018-01-11:
My new 5xC v1-1 Smoothieboard can not establish a connection with bCNC. Python27 shows: Key error: "MPos". I followed all the instructions for use (updated firmware, installed bCNC and other software to communicate with Smoouthieboard, ......). None of these software, which I installed on my Notcob ASUS (SO .- Windows 10), can not establish a connection with Smoothieboard (nor USB, nor Ethernet). How to solve this problem and continue with the installation of Smoothieboard in my CNC machine

eglrd, 2018-01-18:
Same issuer on Mac OS darwin - having followed all install process. Got the board to communicate when using GRBL profile (and not smoothie), though... yet interface was showing "not connected" as a status (my machine was moving ok). No feedback was showing on the screen (coordinates or anything)

reedb, 2018-01-28:
OSX, Python ver: 2.7
smoothieware ver: master-96eb5ef
bCNC ver: 0.9.12
File "/Users/reedb/projects/bCNC/Sender.py", line 1097, in serialIO
    CNC.vars["mx"] = float(d['MPos'][0])
KeyError: 'MPos'

It looks like bCNC is having trouble parsing the following line received from smoothie:
Received line: <Alarm,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.0000>

@eglrd, while trying to get smoothie ware to communicate I saw the same unclear and inconsistent status indications.

reedb, 2018-01-28:
it looks like this commit is causing the parsing issue:

Modify status handling for new smoothie query format

https://github.com/vlachoudis/bCNC/commit/333a257fbadf867759a1ad5f33b1fd7009904a9a#diff-95686cbc55a4e3072b52ec7720dcab71

Perhaps wolfmanjm can provide some guidance on what versions/branches of smoothieware and bCNC work together.

reedb, 2018-01-28:
In my case I had built my own firmware and was on the incorrect branch: master. I changed to edge and was able to connect. You can check this with the version command from a terminal connected to smoothie. 

You also need to have the following in your smoothie config file:

new_status_format                          true               #  talk to bCNC

closed 2018-10-29 by Harvie