vlachoudis/bCNC #721: working coordinates lost

GnomeTek, 2018-01-12:
running windows 10 on a basic commercial off-lease desktop machine on  a shapeoko 3 w/ grbl 1.1 with spindle on/off, and touch probe. 

Working with bCNC to get some more features in my control system. Have been working on using the autolevel feature to get better PCBs over the last several days. I have cut 1 pcb so far, and it was wretched. so it's a work in progress. 

This issue ticket is for _sometimes_ it seems that the system looses the working coordinates after i select autoprobe. I am in my 0,0,0 position, or maybe 0,0,1 to start auto probe. Then It errors out and my coordinates are way out there, my Z comes out larger then the max Z travel. It just freaks. It seems to have happened at other times also, I haven't narrowed down what exactly the glitch is, but the symptom is reference coordinates are lost and I've slammed my carriage into the bed many times because of this. In the 8 months i've owned my machine i have never had as many issues as i do with bCNC. 

outside of many many errors, I'm finding the software very adhoc and not robust. It seems if you do not follow a procedure *just so* then something goes awry. Further, i wish i had more insight into how these procedures are working so i can better understand whats needed from me. For instance with autolevel, why must we select "zero" after we map level? Does Z need to be at 0.0 when we do this? Should we reprobe zero if that is the case?

It sucks to have someone complain about something you've made, i get that. I've written complex software that was very difficult to keep straight. But something i picked up on early was to bulletproof the functions i did have in place. If a user botched something, it was probably a real error and not because they didn't tip toe through the program just so to get the functionality to setup and work correctly. This also total buggers your learning curve and product adoption plummets because of it. There's great features in here that I would love to be able to use, but i'm running dinky little PCB milling bits, i can't afford to have my Z nose dive through the floor. 

Am I the only one? Surely I can't be?

Poly3D, 2018-01-12:
I have had similar issue, I found that avoiding Stop button helps keeping the coordinates,I always make sure the coordinates have not freaked before I start the job.
I crashed many fragile endmills as well, and those are not cheap since I am using T-tech tools which are approx $25-30 a pop. I didn't have any issues with auto level feature yet.

Another issue I have is that the stream just pauses by itself and I have to click "terminal" or "stop pendant" buttons very wierd and totally unrelated, but it helps to get it going.

GnomeTek, 2018-01-12:
I also think it would be very helpful to have "stickied" machine controls. Or atleast the "play, pause, stop" controls available across all tabs. Move it up a level instead of interior to the tab. Just a thought. 

I really want to express that i think you've got a very capable system here. The functionality is there, but i feel like it's not all brought together well. I think that would make it much more user friendly and really make this a great bit of software.

Guenni75, 2018-01-13:
Unfortunately lots of Problems just appear, because there is no description or it is incomplete.
I have not used many things yet because I do not have time to try it until it finally works. In addition, it can be very expensive.
Often the wiki is not really helpfull, too.

A great Piece of Software with much more work to do!

vlachoudis, 2018-01-29:
That's why it is open source so anyone can contribute, making more robust the functions or the documentation.

silverfoxvt, 2018-02-21:
I can confirm that the working coordinates are also lost running bCNC on a RPi if the stop button is pressed during an milling job.  Re-zeroing the work coordinates is then necessary before proceeding.

vlachoudis, 2018-02-21:
@silverfoxvt when the stop button is pressed, bCNC sends a feed hold and a soft reset to Grbl, since is the only way to clean up any command pending in the buffer. Normally the soft reset should not lose the coordinates, at least it doesn't happen in my case with the latest grbl.

einencool, 2018-05-22:
Hello, 
I‘ve the same issues with my machine. 
I‘ve tried a Arduino Uno and now a Arduino Mega. With Grbl 0.9 and now 1.1. The Bcnc Version is from end of april or begin of may.
The Problem loosing the coordinates exist, when I zero my machine with a macro (Z-Touch Probe) and when I Zero my X + Y Axis with a Hotkey on a num-pad. Or when I enter the coordinates in the command prompt.
When I start a job and have to press the Stop or Reset Button, all the coordinates are lost and replaced by some „strange“ coordinates . (But this only happens one time when I press Stop or Reset)

When I Zero my Axis with the Software-buttons (X=0 / Y=0 / Z=0) the descripted problem will not accure.

My Workaround is the following procedure:
Zero all my Axis desripted above (Touch-Probe etc...)
Then I go to 0,0,30 and press the „Reset“ or Stop Button. Then all my coordinates are lost, but then I enter the coordinates into the boxes in the software. When I will click on reset or stop the next time, my coordinates will remain.

But it would be very nice, when this bug could be fixed, because I‘m still learning and make a lot mistakes, so I have to push the stop-button and in the worst case all my coordinates are gone..

Greets 
Chris

silverfoxvt, 2018-07-16:
I stopped using bCNC months ago because of this bug.  It's a pita to deal with for the very reason einencool mentioned.  Just stopping back by on the outside chance it had been fixed, but I see that it hasn't.  Very cool program otherwise.

einencool, 2018-07-16:
Hi Silverfoxvt,

I've changed my zeroing routine from "G92 x0y0" -> "G10 P1 L20 X0 Y0"
Now it works really fine, and the Zero values are stored every time.

Hope this helps you, for me it wasn't easy to find this commands, but now I have no Problems with my values...

Greets
Chris

Harvie, 2018-10-29:
Do you still have these issues?

einencool, 2018-10-30:
Well, since the change of the zeroing routine, it works like a charm 
Greets Chris

Harvie, 2018-10-30:
I've found this: https://machmotion.com/cnc-info/g-code.html#G92,G92.1,G92.2_&_G92.3_Offsets

> G92,G92.1,G92.2 & G92.3 Offsets
> 
> See the chapter on coordinate systems for full details. You are strongly advised not to use this legacy feature on any axis where there is another offset applied.

I am not really sure how G92 and G10 works. Do you think bCNC should not use G92 at all?

einencool, 2018-10-30:
When I set zero on the machine with G92 x0 y0, and then I press the reset button, my coordinates are lost.

With the new routine it works very well.

If I should use it, I don't know, but it works :-)
Greets Chris

Harvie, 2018-10-30:
I would like to get this improvement to upstream bCNC if it makes sense. But i still don't understand the difference. I will probably need to read something about different offsets available in GRBL.

PannDemic, 2018-12-07:
G92 as I understand it is a temp coordinate set. Hence the loss on a reset. G10 I am still looking into. Though this effects bCNC on my RPI as well.

silverfoxvt, 2018-12-12:
Replacing G92 with G10 P1 L20 works to store the coordinates when zeroing even after emergency stop on RPi for me as well.  Thanks @einencool

Harvie, 2018-12-13:
So i guess this is not a real issue and we should close it?

silverfoxvt, 2018-12-14:
From the perspective that the program is keeping the coordinates after clicking stop using the alternate G10 code on the RPi, I would say yes.  The only comment I would make with regards to this though is that in previous versions of bCNC, the G92 command worked in this fashion after clicking the stop button.  It stopped working after the upgrade to the current version even though by apparent explanation of the G92 command, it should have worked in this manner all along.  That's my observation at least.

Harvie, 2018-12-14:
> by apparent explanation of the G92 command, it should have worked in this manner all along.

Yes. I think the same.
If somebody disagrees with this, please comment or reopen the issue.

closed 2018-12-14 by Harvie