vlachoudis/bCNC #723: Can't open GCode file produced by CamBam

bjoerngiesler, 2018-01-16:
I've attached a GCode file produced by CamBam. It consists of two pockets and six profile operations. bCNC opens this file but only displays, and performs, the first pocket operation. If I split the whole thing into separate files, bCNC only opens and displays the first correctly. UniversalGCodeSender has no such trouble, it opens, performs, and visualizes the whole file.

[Gewürzhalter.Part1.nc.zip](https://github.com/vlachoudis/bCNC/files/1635772/Gewurzhalter.Part1.nc.zip)

onekk, 2018-01-17:
Maybe it is too  long it's a 1.9mb file with nearly 60.000 lines of gcode.

It will machine correctly and the problem is related only to the visualisation part?
For the visualisation part there are some post about increasing a variable to make more operation to be displayed correctly in bCNC.

What CamBam post processor have you used?

I  regularly use CamBam with bCNC but generally my file are smaller than yours.

Regards

Carlo D.

bjoerngiesler, 2018-01-17:
TBH, I have not tried to machine it with bCNC. Would have taken too long. But it's not really a very complicated part. 2MB of GCode is not a surprising amount, in my opinion.
Post processor is the default one.

onekk, 2018-01-17:
You have to use GRBL post processor and not the default using bCNC, it is not included by default but if search on CamBam forum you will find it  at least two:
one is the standard one,
the other is modified by me (same nick on the Cambam Forum).

Sorry but I don't remember the exact post now.

bjoerngiesler, 2018-01-17:
Can't find it. But the GCode produced by the default processor works great for me so far; I only have to erase the G64 command in the header which isn't supported by GRBL, and that's only to appease bCNC. UniversalGCodeSender will send the whole thing just fine.

onekk, 2018-01-17:
Here the post processor that i use

[GRBL.cbpp.txt](https://github.com/vlachoudis/bCNC/files/1639951/GRBL.cbpp.txt)
rename it to GRBL.cbpp and put in ~/.config/CamBam plus 0.9.8/post if you are on Linux elsewhere check where this file is put on windows (I don't use windows my win10 machines are very slow after some windows updates).

Regards

Carlo D.

vlachoudis, 2018-01-29:
@bjoerngiesler bCNC is very picky on the errors return by GRBL. If you want to omit a certain character, check the wiki to add the G64 as a command not to be send to grbl

closed 2018-01-29 by vlachoudis