vlachoudis/bCNC #724: Spindle button did not start spindle using raspbian

wwjohb, 2018-01-17:
When using bCNC on raspberry pi running raspbian, the spindle button does not start the spindle, typing the GCode (m3) in the command did turn the spindle on.
I tracked it down to case comparison in the following code:
diff --git a/ControlPage.py b/ControlPage.py
index b62da78..fe4cffc 100644
--- a/ControlPage.py
+++ b/ControlPage.py

        def spindleControl(self, event=None):
                if self._gUpdate: return
                # Avoid sending commands before unlocking
                if CNC.vars["state"].upper() in (Sender.CONNECTED, Sender.NOT_CONNECTED):
                        return
                if self.spindle.get():
                        self.sendGCode("M3 S%d"%(self.spindleSpeed.get()))
                else:

so I just added an .upper() call to the array entry it was comparing the results with and then the button worked again. Should be compatible across both Linux and Windows variants.

Hope that helps.

vlachoudis, 2018-01-29:
I am wondering how it can work the Sender.CONNECTED is "Connected" converting it to
upper case it will never match.
Like this you are sending the command despite the status of the machine.
Did you try to "unlock" first your machine before clicking the spindle to start?

wwjohb, 2018-02-01:
What happened was that the comparison always failed and CONNECTED was never matched , even though the machine was connected. So the function always just "return".  When I was printing the variable out to stdout, it was printing the CNC.vars["state"] variable strings out as "Connected" so in a case sensitive OS  like Linux that fails to match. All I did was make the check match what you were checking for in the array. Either CONNECTED or NOT_CONNECTED. if you specify them with all uppercase letters that is what has to be tested for. 
There was no alarm condition, it was not in a locked state. Spindle just never turned on unless I manually typed the M3 command in "Command" box at the bottom of the screen of the app.
fixing the comparison fixed the button in my case. Now works great.

Hope that helps.

closed 2018-04-12 by wwjohb