vlachoudis/bCNC #727: Autolevel confusion

dmon1234, 2018-01-24:
Hi,

This is not a bug per se but more of a clarification.  I have the autolevel working fine.  My question is about setting the autolevel to 0.  In the wiki, it says that set zero associates the autolevel map to x,y position.  In the PCB 101 it seems that after the autolevel routine ends, we zero the cutter with the probe routine, then set zero in autolevel...  Which way is correct?

I set the cutter to 0 with the probe.  Moved z=1. Then I performed the autolevel routine.  When it ended, the cutter moved to x=0,y=0.  I then moved z=0.  Visual check shows the tip of the cutter exactly touching the metal surface.  I THEN did the autolevel 0.

Is this okay?

To be frank, the pcb cut was awful :-(  Much deeper than I expected...  I wonder if it could be caused by me setting the autolevel incorrectly...

Many thanks
D

gibo77, 2018-01-28:
Can I save the autolevel data in a file so I can use it in the future? When I open the old file that has autolevel data, the auto level data do not show up. If you have numerous passes and you want to remember the autolevel data, how can I do that? 

I goes the only solution is to use bigger tool size bit, so you can do all the passes in one sitting.

gibo77, 2018-01-28:
dmon1234 after you probe zero, next to do is  autolevel. The x=0y=0 point is the z=0. So the next autolevel probing point is with respect to x=0y=0. So for example the next autolevel data is z=0.5, it means that the surface is higher by 0.5 compared to the x0y0.

dmon1234, 2018-02-12:
Hi gibo77,

Apologies for the very late reply.  Do you mean that I should start the autoprobe with the tool at z0? Then after probing, go back to x0,y0 (where the autoprobe started), make sure the tool is at z0 and then hit autoprobe zero?

I did a test run without autoprobing and had surprisingly good results, ie, my cnc was capable of very accurate and shallow cuts (0.05 mm depth of cut).  I would love to get this to work properly for me...

D

goddisignz, 2018-02-12:
Hi,
the routine I run is:

1. Go To X0 Y0
2. Probe Z
3. (Set Z-position to 0)
4. Move Z up (maybe to 0.5mm)
5. Autoprobe
6. Run

I do not use the autoprobe zero as I found out it is not necessary. Especially when you zeroed at X0 Y0. You can use the autobrobe zero after autoprobing to adapt the reference point and height.
But even if you do not do step 3 the autoprobing information is correctly used to compensate the height differences.

dmon1234, 2018-02-12:
Hi,

Thanks so much for taking the time to reply.  I will try the above and let you know.  Do you know if there is a way to see the new adjusted Z which is applied to the gcode, as gcode?  I know that bCNC does the adjustment on the fly but is there a way to see the adjustments?  I guess that I can always look the Z display....

Regards
D

goddisignz, 2018-02-12:
As far as I  know, when you open a new file after autoprobing, bCNC asks if you want to store the autolevel information. It is not the information itself but the complete height compensated gcode. I'm not sure if you need to start the gcode before or not. Just give it a try.

Harvie, 2018-10-29:
Please refer to wiki. If unsure after reading that, you can ask more here.

closed 2018-10-29 by Harvie