vlachoudis/bCNC #728: NameError: global name 'base' is not defined

Bodagous, 2018-01-26:
Forgive me if this has already been addressed, I searched the issues here and did not find a similar one.  

So im unable to really pinpoint what is causing it because it happens almost randomly and the only thing i can do to combat it is too reboot my pc which most of the time cures it.  but im running MX-linux 16 and Python 2.7.9 and starting bCNC from the terminal.  The program flashes breifly on my screen just as if its going to start up like normal however it does not and im looking at the terminal again with this error given.

Traceback (most recent call last):
  File "./bCNC.py", line 2536, in <module>
    application = Application(tk)
  File "./bCNC.py", line 195, in __init__
    page = cls(self.ribbon, self)
  File "/home/outside/Downloads/bCNC/CNCRibbon.py", line 106, in __init__
    self.register()
  File "/home/outside/Downloads/bCNC/FilePage.py", line 314, in register
    (SerialFrame,))
  File "/home/outside/Downloads/bCNC/CNCRibbon.py", line 125, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/outside/Downloads/bCNC/FilePage.py", line 235, in __init__
    devices = sorted([x[0] for x in comports()])
  File "/usr/lib/python2.7/dist-packages/serial/tools/list_ports_posix.py", line 122, in comports
    return [(d, describe(d), hwinfo(d)) for d in devices]
  File "/usr/lib/python2.7/dist-packages/serial/tools/list_ports_posix.py", line 93, in describe
    return usb_lsusb_string(sys_usb)
  File "/usr/lib/python2.7/dist-packages/serial/tools/list_ports_posix.py", line 80, in usb_lsusb_string
    return base
NameError: global name 'base' is not defined

vlachoudis, 2018-01-29:
It seems like a python installation problem. Could you remove and re-install python

Bodagous, 2018-02-14:
tried reinstall and it didnt work but Ive pinpointed the issue too only happening when the controller in this case an uno with generic cnc sheild is plugged into the computer while the computer boots up from being off.  If i plug in the controller only after reaching the desktop and than bcnc works all the time.  So it seems this is not the proper place be looking for answers to this problem, feel free to close this issue unless you have any idea about this..  but i thankyou either way good sir!  and im off too hunt elsewhere for an answer.

closed 2018-03-14 by Bodagous