vlachoudis/bCNC #732: Rotating G CODE

fusibiu, 2018-01-28:
Hi,

Please, try rotate the attached gcode file. Nasty results, isn't it? On my opinion this relates to the missing X or Y coordinates in each block but, unfortunately, I cannot check related bcnc.py code. Could you help?
Note: the code was generated by HSMXpress freeware software and my hobby CNC machine (foreseen with Arduino & grbl firmware) had accept it.

Thanks!

[Flat Part.txt](https://github.com/vlachoudis/bCNC/files/1670884/Flat.Part.txt)

fusibiu, 2018-01-28:
Hi,
It's me again. Sorry to bother you but now I'm sure what's the problem. bcnc needs a fully defined block (operation gcode and coordinates) and do not "inherit" the missing data from the previous block. Please check the attached files: 
    1. Test2 is a striped version of a gcode generated by HSMXpress;
    2. Test2 Completed is the hand edited version of Test2 where the blocks had been completed with the "missing" data. This gcode may be rotated correctly!

Thanks vlachoudis for the future code modification/adaptation.

Regards!
[Test2.zip](https://github.com/vlachoudis/bCNC/files/1670920/Test2.zip)
[Test2 Completed.zip](https://github.com/vlachoudis/bCNC/files/1670921/Test2.Completed.zip)

fusibiu, 2018-01-28:
Hi,

I'm like a politician: I don't know how to do it myself but I can advice you...

So, this is my advice: bcnc should record after each block processing the LAST operation gcode and coordinates (the coordinates should be the start point coordinates in the beginnig). These data should be used in next block processing if the block is not fully defined ("not complete").

vlachoudis, 2018-01-29:
Thanks. This was an old issue that I thought I fixed it, but maybe not for every case. I will check again

vlachoudis, 2018-01-29:
Ok, I found and corrected the problem. The rotate function was not reseting the arc offsets after the command. Now the new master should be ok

closed 2018-01-29 by vlachoudis