vlachoudis/bCNC #734: Problem with Arduino Mega and BCNC

AlvaroRey, 2018-01-29:
Hi guys I see a post realated to this, but I'm builng a machine, and I trying to use mega 2560
I'm using Python 2.7 and Piserial 2.7. The bcnc when I try to connect Show Not Conected status, but if you click over, it said that the connection was stablish with grbl.

If I plug a Arduino Uno, everything works fine, but I can't do anything with mega

Any ideas??
Thanks in advance

goddisignz, 2018-01-30:
Did you program the mega board with the current grbl firmware and have you checked, that the device you try to open is the right one?
What does "dmesg|tail" show right after you plug the mega board?
Have you tried connecting without bCNC, for example picocom or simply tail (tail -f /dev/<yourdev>)?

AlvaroRey, 2018-01-30:
Yes I upload GRBL form Mega, the computer pics the com without problem, I even try all the diferent baudrate and all have the same response Not Connected
I'm using windows, not linux.

goddisignz, 2018-01-30:
Under windows you could use putty instead of picocom. The baudrate should be 115200. After connecting, the mega board should answer with something like "GRBL v1.1". If not, it is probably not correctly programmed.

closed 2018-10-29 by Harvie