vlachoudis/bCNC #735: error:20

kaizoku-619, 2018-01-30:
Hey guys, when I try to start running the program I get the following error: **error:20** the error indicates that an unsupported or invalid g-code found in block. But when I try the g-code in another cnc software it works fine.
[image]

vlachoudis, 2018-01-31:
Click on the red "error:20" button to see which line is causing the problem.
It is mentioned several times, that bCNC stops for safety when any error is reported from the controller, while other senders silently continue.
However there are ways (see the wiki) on how to tell bCNC to not send any "invalid command" from the controller.

AhmedGhar, 2018-02-01:
@vlachoudis , I have the same problem, and I couldn't setup bCNC t not send any "invalid command".

Harvie, 2018-10-29:
Should be fixed. Please use latest grbl and latest bcnc git master. reopen if still needed.

closed 2018-10-29 by Harvie