vlachoudis/bCNC #737: Probe Orient creating g-code errors

gattarin, 2018-02-03:
Hi!

I'm trying align my stock using Probe->Orient. This used to work well, but today I think I stepped on a bug.

When trying to orient attached file, bCNC will change this line:
G18 G3 X44.314 Z0.5 I-0.317
to
G3 G3 X44.287233 Z0.5 I-0.316973 Y-8.523884 J0.00414

This line is failing to run on my GRBL 1.1f.20170801

Is this a BUG?
If you could advise workaround I would much appreciate it?

Thanks,
Marcin

[Holes v1.txt](https://github.com/vlachoudis/bCNC/files/1692155/Holes.v1.txt)

gattarin, 2018-02-05:
> Thank you for the bug report. I will try to fix it

Sure! This is least I could do in return for this great software!
Thanks!
Marcin

vlachoudis, 2018-02-05:
Please try the new master. It should correct the problem

closed 2018-02-05 by vlachoudis