vlachoudis/bCNC #741: GUI Freezes on startup

RaphaelBruegger, 2018-02-06:
Hi Guys 

So when clicking on bCNC.bat or starty bCNC.py from the command line the GUI opens but freezes before loading the menu.. Any ideas what I might be doing wrong? There are no errors shown whatsoever..
[image]

I'm using the newest version of bCNC straight from the git repository..

Thanks alot for your help!

vlachoudis, 2018-02-07:
Try to launch from a terminal and see if python writes any error message

RaphaelBruegger, 2018-02-07:
I tried and got an empty terminal.. No errors..
[image]

Harvie, 2018-10-30:
try latest master pls