vlachoudis/bCNC #744: bcnc is opening then directly close

BesherV, 2018-02-09:
I already download bcnc and also UGS and in both of them I have the same problem 
when I open the program it just 2 or 3 seconds then it is closed 
I already tried to open it using cmd 
I delete python stuff and reinstall it and nothing change 
am so confused because I don't know why I have such a problem with both programs especially that three month ago it was working properly 
please help :)

vlachoudis, 2018-02-12:
We need more information to help. Any error message in the terminal?

BesherV, 2018-02-13:
actually the problem is there is nothing more than that no error message 
I just click on the program first it opens the black window then it opens bcnc then it is directly closed no error nothing else

vlachoudis, 2018-02-14:
This might point to a broken python installation

BesherV, 2018-02-14:
The point is I tried bcnc and also UGS it is always the same mistake. and I already reinstalled python package again and it keep making the same problem that is why I don't know what is the problem. am thinking about formatting the computer itself.

vlachoudis, 2018-02-14:
What OS you are running on?

BesherV, 2018-02-14:
windows 10

vlachoudis, 2018-02-14:
Could you re-install the drivers for your controller

BesherV, 2018-02-14:
which drivers ?

vlachoudis, 2018-02-15:
I am not a Windows user, but I presume that when you have installed the arduino software drivers

BesherV, 2018-02-18:
problem is solved I format my laptop and now it is working I think that there was some incompatibility between windows version and other programming compilers (java and python) due to some maleware affected my CPU 
thanks a lot