vlachoudis/bCNC #745: Z-axis increment during probe / autolevel

Lester444, 2018-02-12:
Is it possible to change the z-axis increment when doing probe / autolevel?

I am aware that I can change the RATE of jog with the "Probe Feed". But this is rate (distance per unit time). I hope to get more accuracy during probing and the z-axis increment seems to be fixed at ______ (0.05mm? not sure).  Is it possible to change the z-axis increment somewhere in the files?

Thank you for your wonderful software!!!  I have been using it for almost 1 year now. Thanks!!!  :)

goddisignz, 2018-02-13:
Hi Lester,
i guess by increment you mean the steps you need to go to travel a given distance. The values for each axis can be found in the controller config variables $100 (X), $101 (Y), and $102 (Z) and are given in steps per mm.

The steps per mm are calculated by:

- the number of (full) steps your stepper motor has per revolution (usually 200, sometimes 400)
- the microstep settings of your stepper drivers (1, 2, 4, ... , 256 - the max depends on your driver board)
- the steepness of your threaded spindle (e.g. 8mm per rev. for a TR8x8 spindle)

The steps per mm are calculated by: num_steps*micro_stepping/steepness
My motors have 200 steps per rev, I use a DRV8825 with a microstepping of 32 and the TR8x8 spindle giving me 800 steps/mm or 0.00125 mm per step. With the A4988 the maximum microstepping is 16 so the steps/mm would be 400.
Both numbers appear quite accurate for a home made or non-professional milling machine.

The only parameter you **could** adjust via software is the microstepping. So you would need to change the logic level of the config pins of your stepper drivers using the grbl controller.
This is not impossible but as far as I know not implemented because usually the config pins are set via dip-switches or jumpers. Of course this also depends on your "main" board.

I do not know settings, but here is my "advice". Just set the microstepping (only) for your Z-axis to the max and adapt the $102 value to match your setup. Of course the has an influence on the maximum travel speed but I would assume that it is still sufficient.

Lester444, 2018-02-15:
This is great!  For the Z-axis I now have 320 steps/mm, 200 steps/rev motor & 5mm pitch lead screw.

 I will try your advise to change microstepping to arrive at 800 steps/mm, Z-axis only.  Then I will come back here and report my results.

Thank you!  👍

closed 2018-02-15 by vlachoudis