vlachoudis/bCNC #759: Camera not working with bcnc

alhaddar, 2018-02-14:
Hi
The Camera not working with bcnc So I tried to install  Python Projective Camera Model with this cmd 

pip install git+https://github.com/smidm/camera.py.git
But show me this letter 
Error [Error 2] The system cannot find the file specified while executing command git clone -q

Please help explain to me the steps to activate the camera in

Step by step where I am new in this field 

Regards

vlachoudis, 2018-02-15:
bCNC works only with the Cameras that are supported by the opencv library. I don't know if this Projective Camera Model is supported by opencv, I could not find it anywhere in the documentation.

alhaddar, 2018-02-15:
thank for you help

now working with me

Regards

vlachoudis, 2018-02-15:
thanks, i am closing the issue

closed 2018-02-15 by vlachoudis