vlachoudis/bCNC #768: Command has invalid target.

Giochia, 2018-02-19:
Hi, your software is very good, professional and full of features, that I have not yet got round to discover in full. A very good job indeed.
I have encountered a small problem, and I cannot understand what am I doing wrong.
The file runs and, immediately, it stops and I get an error that says "motion command has invalid target G2, G3 and G38.2 ....    the arc is impossible ... or probe target is the current position..."
The line at falut is "G2 X10.45 Y2.50679 I1.564 J0.851".

Thank you, Giorgio

vlachoudis, 2018-02-19:
The message is reported by the controller, who stops the execution.
It seems that the G2 arc has an accuracy error on the ending position.
Only from this line is not possible to understand how much, it needs
also the previous line which has the starting position for the arc.

Giochia, 2018-02-19:
Here you can see some preceding and some following lines; hope it will help.

...
F200
(Block-name: Ant1-1)
(Block-expand: 1)
(Block-enable: 0)
G0 X10.666 Y1.69979
M4
(Ant 1-1)
G1 Z-1
G2 X10.45 Y2.50679 I1.564 J0.851
G1 X10.624 Y3.37279
G2 X10.985 Y3.98079 I2.881 J-1.296
...

Guenni75, 2018-02-24:
Hi. 
GRBL has problems for years with G2. 
There are a few forums posting this Problem!

Giochia, 2018-02-24:
The same script runs without problems on other tools. So I tend not to believe it is simply a GRBL issue.

vlachoudis, 2018-02-24:
@Giochia @Guenni75 the error is not in grbl and not in bCNC. The error is in the gcode
the first path has a radius of 1.7805 while the end-point has a distance from the arc-center of
1.934. This is an important error, and grbl rightfully complains with an error. bCNC when it gets
an error from Grbl stops the execution.
Most of the other senders they simply hide the errors and continue the execution. If one hides the errors what is the point of the error message in the first place?

I am thinking of adding a setting on how bCNC to treat the errors, with 3 options:
1. Stop execution on any error (like now)
2. Feed hold/Pause and let the user decide whether to resume or cancel
3. Ignore any error and continue sending.

Giochia, 2018-02-24:
This is a good and professional explanation. It makes bCNC even more precious, thank you !
I have to get round learning more features, as I see there are dozens I yet did not get to explore.
Keep up the good work Vlachoudis !

closed 2018-10-30 by Harvie