vlachoudis/bCNC #771: weird issue : top layer gets ruined (too deep) but bottom layer is fine

phil123456, 2018-03-03:
Hi,

first I grave the bottom layer, then everytime I do the top layer, for some weird reason it is graved too deep but I probe the whole area the same way as for the bottom layer and before any engraving

how could I solve this ?

thanks

masterleo, 2018-04-18:
Hello, could you take pictures and include screen shots please ? What is your OS ? What is the version of BCNC ? We need informations  to solve bugs ;) What is the machine you use ?

phil123456, 2018-10-20:
Windows 7
I cannot find the version I use...just tried the last one, there is an unresolved bug in it (Key error something)
I use a chinese CNC3018 .... I cant take screeenshots right now, but twhile one side is routed correctly, the other one is barely touched by the spindle bit

I tried to reproduce it with a simbple example, and it routed fine.I'll try with a more complex board, if I can reproduce I'll send you all the files I created

regards

MARIOBASZ, 2018-10-20:
It's a weird mistake. Can Autolevel have influence?

phil123456, 2018-10-26:
here are the files and pictures

the second (bottom) tap file must have been updated with the autolevel scan I think, not the first one

[image]
[image]

[bCNC_issue.zip](https://github.com/vlachoudis/bCNC/files/2520434/bCNC_issue.zip)

one clearcly see the issue on the pictures

Harvie, 2018-10-30:
You have to do autolevel for each side separately, every time you mount your PCB to wasteboard.
Also note that if you save the autolevel data under same name as gcode, it will get autoloaded.
Isn't this your problem?

phil123456, 2018-10-30:
I do autolevel for each side, obviously :-)

Harvie, 2018-10-30:
So i don't get it. From what you say bCNC is have no way to know which side is milled.
Maybe your board is flexing and once cut it flexes even much? The woodscrews you use might bend and pre-tension the board. Try to use double sided tape to support it across the whole board. Make sure that there is no debris from previous cutting under the board. Also try to lower the probing speed 10 times to see if makes a difference. Maybe the copper on one side of the board is thicker than on another... Maybe there's some backlash or flex in Z, which gets pretensioned while you do first cut. What kind of spindle do you use? Does it have any play in Z (or any other) axis? (loose bearings, so you can move milling but up and down a bit?) PCB engraving is EXTREMELY sensitive for any deviation in Z and if you have play in your machine, the autoleveling just might not be enough. Maybe the bit vibrates up and down while rotating and during autolevel procedure it is not.

phil123456, 2018-10-31:
> So i don't get it. From what you say bCNC have no way to know which side is milled.

Could'n agree more, makes no sense so far

> Try to use double sided tape to support it across the whole board instead of screws 

Yep, thought about that, I'll give it a try

> Make sure that there is no debris from previous cutting under the board

Checked

> Also try to lower the probing speed 10 times to see if makes a difference

hmm, did'nt know about that, I'll see whe I can find that setting

> What kind of spindle do you use

it's the motor that came with the cnc, there is no z movement on the axle at rest, alredy tried
I also use an er11 for holding the bit, and a fresh carving tip everytime

I'll check the whole montage again, good idea

thanks for all these tips dude ;-)

Harvie, 2018-10-31:
>> Also try to lower the probing speed 10 times to see if makes a difference

> I'll see where I can find that setting

[image]

phil123456, 2018-11-03:
I just remembered that everytime I do the scan, I click "reset" otherwise the spindle does not start, I also have to start the spindle manualy

but when I click start, as the spindle is running, the thing goes back to center while beeing really close to the board, instead of setting the motor at high position first...it creates sometimes a really thin line on the pcb going from the start scan point to the center point

not a big deal at first, but that alone maybe damages the bit before even starting

Harvie, 2018-11-05:
> it creates sometimes a really thin line on the pcb going from the start scan point to the center point

Do you have latest git version of bCNC? I think i've fixed this some time ago. But it all boils down to how you generate your g-code. If you use "cut" operation in bCNC it should not happen. Also be sure to disable other blocks and jog Z up before you hit start.

phil123456, 2018-11-06:
I never use python 3 :-)

Harvie, 2018-11-06:
> I never use python 3 :-)

[image]

what is this then?

phil123456, 2018-11-06:
I dont get it, I am sure I use python 2.7, I'll check when @ home

Harvie, 2018-11-06:
Note that you can have multiple python versions installed. That's not really uncommon.
Also in such case you need to make sure, that you install numpy module to the right python installation using pip2. There were people reporting problems with numpy module missing, but they've installed numpy to python3 instead.

Also note that there's python3 bCNC port on the way. Some stuff is still broken, but there will be transition to python3 in few months.

phil123456, 2018-11-14:
right, so I just tried to cnc a pcb and it's even worst than before
not only the bug is not resolved, after a scan the bit is so close to the pcb it traces a big line
but now even if I lower 0.1 unit the z axis befor starting, it still engraves half of the traces

I am looking for another more decent software

Harvie, 2018-11-14:
> bCNC definetely treats bottom and top layer differently

AFAIK bCNC has no concept of layers. So there's no way it will tell the difference. Last piece of your toolchain that can tell layers is CAM or g-code generator, that you use.

> I am now looking for another more decent software

Please let me know what are the results with other software. I can reccomend you to try UGS or newer UGS platform. If you will get better (or exactly same) results with UGS or similar software it will be great starting point to fixing this bug (whether it lays in bCNC or elsewhere).

Harvie, 2018-11-14:
> after a scan the bit is so close to the pcb it traces a big line

Do you know that you probably should not start right after the scan? You probably should retract and/or rezero. Does your g-code contain Z coordinates? At least something like Z0 at beginning.