vlachoudis/bCNC #773: Can't resume after M0

Harvie, 2018-03-06:
When my g-code file contains M00 (eg. for toolchange), grbl pauses, but can't be unpaused from bCNC.
In universal g-code sender i can just click "play" button to continue, while bCNC has "play" button greyed out by the time code reaches M0, so i can only abort the g-code execution by "stop" button once it gets to M0.

goddisignz, 2018-03-07:
Hi Harvie,
for me the "pause" button works as a continue. I usually use it during the tool change procedure (M6).
Have you tried that?

Harvie, 2018-03-07:
@goddisignz According to tooltip it should resume, but it does not work:

[image]

Harvie, 2018-08-09:
@neilgill wrote:

> I tried double clicking Pause, then double clicking Pause again, wait between each double click, after two or three goes, my machine resumed. It seemed to work consistently, several double clicks gets it going.

@vlachoudis @chamnit any ideas why grbl does not resume at first try?

Harvie, 2018-08-10:
I've just tried what @neilgill said and it's true. When feed hold is activated for the first time i need to doubleclick the pause button several times to unpause. Then it starts working as supposed on single click until i restart bCNC.

Harvie, 2018-08-10:
OK. it's not grbl bug. I've hit reset button on arduino and it remained working with single click. Also i've used interceptty to sniff serial traffic and i've found that bCNC is sending "!" instead of "~". Double click probably quickly flips state of **sender._pause** which works around the issue.

Harvie, 2018-08-10:
I've just found in Sender.py on line 1018 if i change:

```
                                                if not self._alarm:
                                                           CNC.vars["state"] = fields[0]
```

To

```
                                                CNC.vars["state"] = fields[0]
```

It works properly

mrdunk, 2018-08-26:
I used to see this issue after pressing the Pause button. (I don't remember trying M0 before.)

I'm currently unable to reproduce the issue with either M0 or the Pause button using a fairly up-to-date grbl version and the latest bCNC from the git repo `master` branch.

Looking at the code currently in `master`, this fix is currently in place for this which explains why things work now:
https://github.com/vlachoudis/bCNC/commit/129ba698a9ed3a4e5893ec3324edb890c3d24c74
```
1017                                                 #FIXME: not sure why this was here, but it was breaking stuff                                                                                            
1018                                                 #(eg.: pause button #773 and status display)
1019                                                 #if not self._alarm:            
1020                                                 CNC.vars["state"] = fields[0]   
```
I have not seen any adverse effects from this change but i'd love to see some proper unit tests for the `Sender` class to verify we can't end up in some undesirable state.

By default it appears the `self._alarm` is set (True) when the serial port opens (in `Sender.open()`)
and needs to be explicitly cleared later.
Looking through the code to see where `self._alarm` gets cleared i see this change:
https://github.com/vlachoudis/bCNC/commit/5f16bbf0c8e49d74af431c42039a35aed10b7e83
`self._alarm`  _used to_ be cleared whenever an "ok" was received from grbl but commit 5f16bbf0c8e49d74af431c42039a35aed10b7e83 changes that behaviour.
Since this implies it is acceptable to continue processing gcode when `self._alarm` is set then i think the fix commit 129ba698a9ed3a4e5893ec3324edb890c3d24c74 is probably safe.

I'm not convinced `self._alarm` actually does _anything_ useful now.
The code only checks for it's state in 2 other places and both those imply grbl is sending us valid information.
Even if `self._alarm` is needed, it is certainly no longer well named.

I'm testing with an Arduino Nano (with no CNC hardware attached).
Grbl version:
```
$I
[VER:1.1g.20180614:]
[OPT:V,15,128]
```
bCNC version:
```
$ ./bCNC --help | head -n1
bCNC V0.9.14 [5 Feb 2018]
```

thanks for listening,
dunk.

Harvie, 2018-10-29:
Seems to be fixed

closed 2018-10-29 by Harvie