vlachoudis/bCNC #775: Can not change drive when opening file

lakeroe, 2018-03-15:
Hello,

I'm running bCNC on Windows 7 on drive C:
How is it possible to directly open a g-code file from an attached USB-Stick (e.g. drive F:) ?
Maybe I'm just to blind but I can't find a button to change the drive ...

[image]

Best regards,
lakeroe

FredericG-BE, 2018-03-15:
I have been struggling with the same issue to reach files on a network drive. 
I think you can type, for example "D:\", in the "File Name" field (and press enter)

FredericG-BE, 2018-03-15:
Not sure if there is some control from Python on this dialog to make switching to other drive-letters  more intuitive? That would be good.