vlachoudis/bCNC #776: python pyserial module error (and other issues)

bigshug, 2018-03-16:
Hi folks.

Been struggling with this for a whole day now.  Continually get -  Error: Please install the python pyserial module

Firstly - does bCNC required Python 2.7 as stated in the WIKI? 

Following the WIKI process of installing python results in the latest Python 3.6.4 being installed - not Python 2…

The ‘export PATH’  in the wiki is applicable to Python 3 but in Python 2 the path is wrong.  In Python 2 There is no bin folder in libexec - it’s one level up.

export PATH=$(usr/local/opt/python/libexec/bin:${PATH}

export PATH=$(usr/local/opt/python2/bin:${PATH}

When I install pyserial - I get  - Requirement already up-to-date: pyserial in /usr/local/lib/python2.7/site-packages.  Should this be somewhere else?

The other error is with zlib.  Error: homebrew/dupes was deprecated

So - I seem to be VERY far from getting this working unless some kind soul can help me out…

bigshug, 2018-03-17:
Correct path for Python 2.7 is-  **export PATH="/usr/local/opt/python2/bin:$PATH"**

zirkeltraining, 2018-03-25:
i had the same startup error. I have both python 2.7 and 3.6 on my mac os X (10.11.6. El Capitan). 
when i try to start bCNC with
'python /(path.to)/bCNC.py' I get the error "Please install the python pyserial module" although i did install it with pip for both python versions already.
i need to use 
'python2 /(path.to)/bCNC.py' and then it works.

Harvie, 2018-10-30:
i also have to explicitly use python2 binary. i think it's normal.

closed 2018-10-30 by Harvie