vlachoudis/bCNC #778: Units conversion and Repeatable, Unexpected Deep Plunge Issue

dramsay9, 2018-03-17:
I converted a file using flatcam and tried to mill it.  The first two times it looked great-- the third-sixth times the toolhead plunged through the workpiece and into the bottom of the plate.  On the bCNC screen, it shows the red arrow moving incredibly far and wide-- effectively tracing the correct toolpath, but 5x larger than the path on the screen.

The nondeterministic behavior (whether it works or not) has got me scratching my head.  It's quite bizarre that it worked twice perfectly, and that every time after that it appears to plunge relatively quickly into the workpiece, in nearly the same spot.  It seems like the units are getting switched behind the scenes/at arbitrary times.  I tried pulling the latest version and that didn't change it.  Here's my gcode:

[mill_gcode_1v0.txt](https://github.com/vlachoudis/bCNC/files/1821360/mill_gcode_1v0.txt)

Thanks for the help, this is my first time milling and I've lost quite a few tips!

FredericG-BE, 2018-03-18:
I observed that kind of behavior years ago, when I was using an (electrically) very noisy router. After I had started the router I had to reset grbl or it would often just move quickly in one direction and hit the machine limits. And sometime it would just happen randomly.
IMHO, what you are are seeing is noise that corrupts the grbl state.
If I where you I would try this multiple times without cutter, with and without running router/spindle.

dramsay9, 2018-03-20:
Thanks Frederic!  I played around with moving some wires and grounding and the issue seems to have gone away, though I'm still holding my breath!  I appreciate the guidance!

closed 2018-03-20 by dramsay9