vlachoudis/bCNC #779: Serial not working correctly.

mano1979, 2018-03-18:
When i start bCNC the serial connection is not working correctly. it makes a connection but the status bar still says "not connected", and the cnc is not reacting to any buttons. If i use another program to make a serial connection, all is fine. After that bCNC can also connect properly.

I am using GRBL 0.9 (grbl-for-cyclone (ramps1.4 and arduino mega2560))

jpascher, 2018-03-26:
did you install the serial 3.4 and Python 2.7.4 correctly?  Serial driver has to be installed or added to Python in a  command line  way. The fist time i installed my driver did not work either.

Harvie, 2018-10-30:
Does this still happen? I tried grbl for ramps once and it seemed buggy to me, but i wasn't doing much testing. have you tried real grbl?

PR #1525 (2021-02-18): Fix STATUSPAT and POSPAT for non-6-axes GRBL0 controllers
The optional, non-capturing groups introduced with 5abe8cb98603f84ac57fdd67307e18c2b611796a had capturing groups inside them, causing nulls to be captured, causing:

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/aib/proj/OS/bCNC/bCNC/Sender.py", line 787, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/home/aib/pro [... cut]
closed 2021-02-18 by Harvie