vlachoudis/bCNC #780: bCNC freezing and unable to close.

mano1979, 2018-03-18:
Sometimes bCNC freezes up and i am unable to close it.  First i try the red "close" button in bCNC but that never works. Then i got to the terminal that i used to start bCNC, and press CTRL+C. This kinda works but it never goes back to the command line. After pressing CTRL+Z it is shutdown in the terminal, and no active instance of bCNC can be found. But the bCNC window stays open and i am unable to close it. i have to reboot my pc to get rid of it.

I recently picked up on cnc-ing again, and updated bCNC. But with the older version i used before, i never had this problem.

vlachoudis, 2018-03-18:
Hi @mano1979 could you provide more information which version of bCNC was working, what OS, do you see any messages in the terminal etc.

mano1979, 2018-03-19:
I am on linux (ubuntu 14.04). The previous version i used is unknown. But it was at least 2 years old. O checked for messages in the terminal. Initially i get no messages but on a bad crash, when i stop it with CTRL+Z it throws this: (see next message)

mano1979, 2018-03-19:
`Traceback (most recent call last):
  File "./bCNC.py", line 2530, in <module>
    application = Application(tk)
  File "./bCNC.py", line 459, in __init__
    self.openClose()
  File "./bCNC.py", line 2051, in openClose
    if self.open(device, baudrate):
  File "./bCNC.py", line 2066, in open
    parent=self)
  File "/usr/lib/python2.7/lib-tk/tkMessageBox.py", line 91, in showerror
    return _show(title, message, ERROR, OK, **options)
  File "/usr/lib/python2.7/lib-tk/tkMessageBox.py", line 72, in _show
    res = Message(**options).show()
  File "/usr/lib/python2.7/lib-tk/tkCommonDialog.py", line 48, in show
    s = w.tk.call(self.command, *w._options(self.options))
_tkinter.TclError: can't invoke "grab" command: application has been destroyed
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb545258c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5447e6c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5454e6c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5447a8c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb545226c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d0cc>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5447b4c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d48c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb54541ec>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5452f0c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5454aec>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544dd6c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d18c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5452bac>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb545240c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d8ac>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5452b0c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544792c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb545476c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb545410c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb54527ec>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d9cc>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d1ec>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544dc6c>> ignored
[... cut ...]
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5452aec>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb544d30c>> ignored
Exception _tkinter.TclError: TclError('can\'t invoke "image" command: application has been destroyed',) in <bound method PhotoImage.__del__ of <Tkinter.PhotoImage instance at 0xb5447aec>> ignored
`

jpascher, 2018-03-26:
I have the same problem i am using windows 10 and python 2.7.4 with serial driver 3.4  different errors will be displayed like "hard limit" but one cant get  on. It is summular ti the behaviear as i have if use the setup window and push the final button. I did do the settings via editing the ini file and i don touch the Window for graphical setup. Would be quite happy with it if this Problem would not cores a complete new start and kill of the old instance.

jpascher, 2018-03-26:
Hi i had an intense look into it and the problem is in connection with the communication of the grbl1.1.
I ha a look at the terminal window the fist time one saves a config to the grbl CNC it sends the code but i stops with 25%. To get back one can use the stop send butten. On could sen commandline $$ before but not after. repeating this results also in not sending commands.

Harvie, 2018-10-30:
is this still issue with latest master?