vlachoudis/bCNC #782: bCNC when fist connect to DIY3018 makes noise

luiscabus, 2018-03-20:
Hi, I'm using gbrl v0.9 on woodpecker 3.0 board, DIY3018 chinese hardware. 

When I open bCNC, I chose the port (COM9 usually), correct baud rate, and hit OPEN (on the connection icon). Then my machine briefly turns ON the Laser or Spindle (depends on which is connected) and makes some noise on the stepper motors.

I checked the py but it's clean. But on the console/terminal of bCNC I can see the following: (sometimes it shows more commands than this before the Grbl Welcome Help~)

`88888<<<<<8888888888888888<<<<<<<8888
G43.1Z0.0
Grbl 0.9j ['$' for help]`

Any ideas? Kind Regards

FredericG-BE, 2018-03-20:
Not sure what bCNC exactly does when opening the port, but the Arduino restarts. I suppose this is intentional, this also happens when the Serial Monitor is opened in the Arduino IDE. The restart is probably provoked by togging the UART handshake signals.
Before GRBL starts, the bootloader runs. It is a known issue that the stock Arduino bootloader toggles the D13 pin a few times. On my system this pin is used to enable the spindle and I also have a vacuum cleaner activated with that pin. The spindle controller is too slow to react to the quick pulses but the vacuum cleaner does; it shortly starts 2 times when the Arduino resets (and so also when i connect with bCNC or start the Aduino Serial Monitor)
I assume this is what you also experience. Possibly you have another bootloader on your board that activates other/more pins so that you also hear it in the steppers and see some rubbish (other baudrate?) on the serial port.

vlachoudis, 2018-03-20:
@FredericG-BE when anything connects to the serial port the arduino restarts, it is not bCNC that is forcing this restart.

FredericG-BE, 2018-03-20:
I did some reading
It is the DTR line that causes the restart

>  it is not bCNC that is forcing this restart.
It seems that just opening the port without any options can cause this, [see pyserial issue](https://github.com/pyserial/pyserial/issues/124)

> import serial ser = serial.Serial('COM3', 38400)
> causes the DTR line to change (at least once) so my connected hardware is reset. This does not happen > if the port is opened with another software, HTerm (http://www.der-hammer.info/terminal/) which also > allows individual control of these lines.

When it is specific to pyserial, it could explain:
>  Needless to say that I have tested with other gcode senders like gbrlPanel and grblController ... and it doesn't happen there.

luiscabus, 2018-03-20:
I made a short video of what happens when I hit connect: https://youtu.be/hp_b41JCqls

Harvie, 2018-10-29:
Sounds like HW problem to me. Do you still have this problem? Have you tried different arduino or usb/serial chip?