vlachoudis/bCNC #783: Editor block changes produce duplicated coordinates on line

valentinAlkan, 2018-03-22:
When I try to rotate/mirror my gcode files, bCNC produces an output line that has the same set of coordinates on it twice. This then throws error 25 when trying to run the file. I have been able to reproduce this error on two different machines, running the latest pull of bCNC. This started randomly after being able to rotate/mirror my gcode files with no issues. I didn't update or change bCNC in any way, or even restart it, actually. If you keep rotating/mirroring, it will add another set of coordinates to each line every time you press the button. This happens for any one block or all of them individually if you have them all selected.

[image]
[image]

This is my gcode file, as produced by kicad/flatcam. I haven't changed my workflow in any way between this file and files that did work previously.

```
G21
G90
G94
F10.00
G00 Z1.0000
M03 S10000
G4 P1
G00 X10.1600Y-10.1600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-12.7000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-15.2400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-17.7800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-20.3200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-22.8600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-25.4000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-27.9400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-30.4800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-33.0200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-35.5600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-38.1000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-40.6400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-43.1800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.1600Y-45.7200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-10.1600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-12.7000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-15.2400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-17.7800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-20.3200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-22.8600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-25.4000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-27.9400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-30.4800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-33.0200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-35.5600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-38.1000
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-40.6400
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-43.1800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X35.4600Y-45.7200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X8.8900Y-82.5500
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X11.4300Y-82.5500
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X13.9700Y-82.5500
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X16.5100Y-82.5500
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X7.7800Y-72.6600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X10.3200Y-72.6600
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X31.0200Y-56.8800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X33.5600Y-56.8800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X36.1000Y-56.8800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X36.1000Y-72.1200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X38.6400Y-56.8800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X38.6400Y-72.1200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X41.1800Y-56.8800
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X41.1800Y-72.1200
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X29.0700Y-91.7700
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X34.1500Y-91.7700
G01 Z-2.0000
G01 Z0
G00 Z1.0000
G00 X0.0000Y0.0000
M05
M05
```

valentinAlkan, 2018-03-22:
Anyone who is having issues with this can use the following python script to fix your files:

```
from __future__ import print_function

fname = input("Please enter the file name to fix: ")
with open(fname) as f:
    content = f.readlines()

content = [x.strip("\n") for x in content]
content = [x.split(" ") for x in content]

i = 0
for line in content:
	if line[0] == "G0":
		print(line)
		content[i] = line[:3]
	i += 1
		
for line in content:
	print(line)
	
with open(fname + "_fixed", 'a') as f:
	for line in content:
		f.write(" ".join(line) + '\n')
```

This only works for files that have G0 commands followed only by an X and a Y. If you have a Z move in there as well, you need to change this line:

content[i] = line[:3]

to this:

content[i] = line[:4]

But only for the lines with a Z in them.

masterleo, 2018-03-25:
Some mess for me too !!! Does it affect you when you move the gcode too ? Or not ?

vlachoudis, 2018-03-26:
Thanks for the report. I've just fixed, you can test with the new master

closed 2018-03-26 by vlachoudis

masterleo, 2018-03-26:
Great !!! This worked for me !